espressif/openocd-esp32 #2: Can't step over break instruction

projectgus, 2016-12-16:
If CPU stops on a break instruction (ie breakpoint compiled into the binary), stepping will not proceed past it.

`jump +1` or `set $pc X` will allow you to progress.

igrr, 2017-07-27:
Can't reproduce this anymore, closing for now.

igrr, 2018-01-06:
Reported again in https://esp32.com/viewtopic.php?f=2&t=4146

pdelprat, 2018-01-06:
Hello,

a. ESP-WROVER-KIT v3
b. ESP-IDF v3.1-dev-69-g8688f0ec-dirty 
c. Windows 10 latest release
d. Bar metal, not VM

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/1608900/openocd_log.txt)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/1608901/gdb_log.txt)

King regards,
Pascal

erhankur, 2023-01-17:
Closing due to inactivity.

closed 2023-01-17 by erhankur