espressif/openocd-esp32 #5: ESP-WROVER-KIT JTAG

wixa, 2017-02-19:
Hi guys! Can anybody help me starting debug using ESP-WROVER-KIT JTAG FT2232 ? 

I have manual http://esp-idf.readthedocs.io/en/latest/openocd.html, i do 2 steps. Come to step "Installing OpenOCD". Clone openocd-esp32 repository.  After i need "For compilation of OpenOCD, please refer to the README, README.OSX and README.Windows file in the openocd-esp32 directory. You can skip the make install step if you want.", I see this documents but not anderstand hat i need do. Can anybody help me?

krzychb, 2017-06-01:
Hi @wixa, 

If you are using Linux then try https://espressif.com/sites/default/files/documentation/jtag_debugging_for_esp32_en.pdf that provides extra details on installation in chapter 3.2.

Another option is https://www.matthias-jentsch.de/2017/02/16/on-chip-debugging-on-esp32-with-eclipse-and-openocd/

wixa, 2017-06-20:
Thank you @krzychb

closed 2017-11-23 by igrr