espressif/openocd-esp32 #7: Threads listing screw up

baoshi, 2017-03-26:
I repeatedly run into this issue in various places. e.g,
1. Reset the target board.
2. Start openocd session.
3. Start and connect gdb, then I receive this message in openocd terminal
```
Info : accepting 'gdb' connection on tcp/3333
Info : Auto-detected RTOS: FreeRTOS
Error: Error allocating memory for 1073434772 threads
```
at this moment "info threads" from gdb session gives
`* 1    Remote target     0x00000000 in ?? ()`
and program did not stop running

I could, at gdb session, give command "thread 1" and "c", then upon hit Ctrl-C
I receive
```
Info : esp32.cpu0: Target halted, pc=0x400D1750
Error: FreeRTOS maximum used priority is unreasonably big, not proceeding: 1702187384
```
info threads gives no threads although program is halted
Appreciate any help.

josesalcedo, 2017-04-19:
I was having exactly the same problem, I can't remember exactly what I did to fix it, but I remember I had to change some settings under the esp32.cfg configuration file (notice that the #'s are in a new line, the interpreter gives an error if they are in the same line as uncommented line):
```
#set ESP32_ONLYCPU 1
#Only configure the PRO CPU

#set ESP32_ONLYCPU 2
#Only configure the APP CPU

#set ESP32_RTOS none
#Disable RTOS support

set ESP32_RTOS FreeRTOS
#Force RTOS to be FreeRTOS
```

The other thing I was doing, I was debugging the bootloader ELF file instead of the application ELF file.

Just FYI, I'm using a Raspberry PI 2 to debug the ELF files through network. My actual development machine is under a VM. It took me few days to get it working correctly.

Hope that helps.

krzychb, 2017-06-01:
Hi @baoshi,

I see similar issue. To resolve it I am doing reset of board between your steps 2 and 3.

Please see https://espressif.com/sites/default/files/documentation/jtag_debugging_for_esp32_en.pdf, page 8/16, point 3.

krzychb, 2017-08-27:
Hi @baoshi, now I do not see this issue anymore with current release of OpenOCD.

FayeY, 2018-03-06:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-03-06 by FayeY

nkolban, 2018-03-16:
In case it helps anyone in the future.  I was getting the error:

`Error: FreeRTOS maximum used priority is unreasonably big, not proceeding: <large number>`

and tracked it down (for me) to the fact that I was doing some private linking and had **not** included the following in my linkage preparation for my ELF file:

`-Wl,--undefined=uxTopUsedPriority`

Adding this to the linkage cured me right up.