espressif/openocd-esp32 #9: Build fails in esp108_apptrace.c

MrBuddyCasino, 2017-05-01:
I just pulled and rebuilt (with clean and configure), and got this:

`esp108_apptrace.c:615:37: error: cast from 'uint8_t *' (aka 'unsigned char *') to 'struct apptrace_data_hdr *' increases required alignment from 1 to 2 [-Werror,-Wcast-align]
                                struct apptrace_data_hdr *hdr = (struct apptrace_data_hdr *)&at_cmd_ctx->trax_block_data[i];
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

I'm on macOS 10.12.4, gcc -v says:
`Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.1.0 (clang-802.0.42)
Target: x86_64-apple-darwin16.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin`

cjsm74, 2017-05-01:
Same problem on Raspbian Pixel:
```
esp108_apptrace.c: In function 'esp108_apptrace_poll':
esp108_apptrace.c:615:37: error: cast increases required alignment of target type [-Werror=cast-align]
     struct apptrace_data_hdr *hdr = (struct apptrace_data_hdr *)&at_cmd_ctx->trax_block_data[i];
                                     ^
esp108_apptrace.c:630:80: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
       if (write(at_cmd_ctx->fout, (uint8_t *)(hdr + 1) + wr_idx, wr_chunk_len) != wr_chunk_len) {
```

MrBuddyCasino, 2017-05-02:
Can confirm it compiles using: `make CFLAGS+=-Wno-cast-align`

igrr, 2017-06-07:
Fixed in https://github.com/espressif/openocd-esp32/commit/2baa81c34ba00c33f1ec28167161ca3107c7cca9.

closed 2017-06-07 by igrr