espressif/openocd-esp32 #10: Error: Error allocating memory for (billion+) threads

gojimmypi, 2017-05-22:
A few days ago, I was able to get the `xtensa-esp32-elf-gdb -tui` working to single-step ESP32 code! :)

Since that time however, I have been unable to duplicate that success. 

I am running espressif openocd from my `~/workspace/ESP32/openocd/scripts`, and I am using the "hello world" freeRTOS sample that is installed here: `~/esp/hello_world`, following the instructions from here: [http://esp-idf.readthedocs.io/en/latest/get-started/index.html](http://esp-idf.readthedocs.io/en/latest/get-started/index.html)

I'm running the openocd server like this:

```
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
export IDF_PATH=~/esp/esp-idf

cd ~/workspace/ESP32/openocd/scripts
sudo openocd -f interface/ftdi/olimex-arm-usb-ocd-h-1MHz.cfg -c "transport select jtag"  -f target/ESP32.cfg
```
and then I launch gdb like this:

```
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
export IDF_PATH=~/esp/esp-idf

cd ~/esp/hello_world/build
xtensa-esp32-elf-gdb -tui -x hello_world.gdb hello-world.elf
```
The gdb session connects, but openocd complains: `Error allocating memory for 1073431524 threads`; I assume there are not really that many threads running ;)

```
Open On-Chip Debugger 0.10.0-dev-g372bb59 (2017-05-20-10:35)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
jtag
force hard breakpoints
Info : clock speed 1000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32.cpu0: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : accepting 'gdb' connection on tcp/3333
Info : Auto-detected RTOS: FreeRTOS
Error: Error allocating memory for 1073431524 threads
Info : dropped 'gdb' connection

```
I have a custom `olimex-arm-usb-ocd-h-1MHz.cfg` only to add the `adapter_khz` setting:

```
#
# Olimex ARM-USB-OCD-H
#
# http://www.olimex.com/dev/arm-usb-ocd-h.html
#

interface ftdi
ftdi_device_desc "Olimex OpenOCD JTAG ARM-USB-OCD-H"
ftdi_vid_pid 0x15ba 0x002b

ftdi_layout_init 0x0908 0x0b1b
ftdi_layout_signal nSRST -oe 0x0200
ftdi_layout_signal nTRST -data 0x0100
ftdi_layout_signal LED -data 0x0800

adapter_khz 1000
```
the only line in `hello_world.gdb` is this:
```
target extended-remote localhost:3333
```
I've also tried
```
target remote localhost:3333
```
...with the same results regarding the crazy number of threads. I think the only thing that's really *changed* is `apt-get update`.  I've tried this on two different (Ubuntu in a VM) machines.

Per a suggestion on twitter, I've also set FreeRTOS to use only one core via `make menuconfig`:

```
#
# FreeRTOS
#
CONFIG_FREERTOS_UNICORE=y
CONFIG_FREERTOS_CORETIMER_0=y
# CONFIG_FREERTOS_CORETIMER_1 is not set
CONFIG_FREERTOS_HZ=100
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=y
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_NONE is not set
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_PTRVAL is not set
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY=y
# CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK is not set
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=1
CONFIG_FREERTOS_ASSERT_FAIL_ABORT=y
# CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE is not set
# CONFIG_FREERTOS_ASSERT_DISABLE is not set
CONFIG_FREERTOS_BREAK_ON_SCHEDULER_START_JTAG=y
# CONFIG_ENABLE_MEMORY_DEBUG is not set
CONFIG_FREERTOS_ISR_STACKSIZE=1536
# CONFIG_FREERTOS_LEGACY_HOOKS is not set
CONFIG_FREERTOS_MAX_TASK_NAME_LEN=16
# CONFIG_SUPPORT_STATIC_ALLOCATION is not set
CONFIG_TIMER_TASK_PRIORITY=1
CONFIG_TIMER_TASK_STACK_DEPTH=2048
CONFIG_TIMER_QUEUE_LENGTH=10
# CONFIG_FREERTOS_DEBUG_INTERNALS is not set

```

So am I missing something, or is there a known issue with OpenOCD? I hear a new version is coming soon.... :)

krzychb, 2017-08-27:
Hi @gojimmypi,
There have been several updates to OpenOCD since May 22 and I assume this issue is now gone with current release on master :smile:
Could you check / confirm that?

FayeY, 2018-03-06:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-03-06 by FayeY