espressif/openocd-esp32 #12: Debugging is not working with firmware from Arduino IDE

valeros, 2017-06-10:
Hi! I'm having a strange problem with debugging a simple example from Arduino IDE, whereas "Hello World" example from ESP-IDF can be debugged perfectly.
Hardware:  FT2232H + esp32 dev board connected via JTAG.

When I'm trying to debug a simple "WiFiScan" sketch from Arduino in GDB I see either this [GDB log](https://pastebin.com/2WSK9tZj) with next [openOCD log](https://pastebin.com/ERctay12) (looks like MCU stuck in ResetVector) or [this one](https://pastebin.com/Jp1QM8eF) and this [openOCD log](https://pastebin.com/p6Bc2sNg) with a strange thread number.

gdbinit it the same for both cases:
```
target remote :3333
mon reset halt
hb app_main
x $a1=0
c
```
Am I missing something? Any advice and suggestions will be greatly appreciated!

igrr, 2017-06-10:
Could you please attach the elf file, GDB remote log, and openocd log with increased log level (`-d 3`)? Thanks.

valeros, 2017-06-10:
Hi @igrr ! Sure, [openocd](https://pastebin.com/GrJjSEA3), [GDB](https://pastebin.com/8z4K0n5d), [elf](https://drp.io/f/By1d4RKf-)

igrr, 2017-06-11:
Ok, thanks. I've tried reproducing this, looks like there are three issues:
- the ELF file doesn't include `uxTopUsedPriority` symbol, which must be linked in via an LD flag ([source](https://github.com/espressif/esp-idf/blob/master/components/freertos/component.mk#L5)). Maybe the flag is missing in Arduino platform.txt.
- `mon reset` doesn't work reliably if 240MHz CPU frequency is selected. Since we have lowered the default in IDF to 160MHz, I hadn't noticed this previously. Hopefully this is easy to fix...
- Placing first breakpoint on a task which runs on APP CPU doesn't work: breakpoint doesn't get hit because APP CPU is reset one extra time by PRO CPU during startup... this also clears the breakpoint register. I'm not sure if this easy to fix or not, need to poke around a bit.
Overall, thanks for the report, will keep you updated.

igrr, 2017-06-27:
The 2nd point (mon reset issue) has now been fixed, merge pending.
The 3rd issue has been fixed in ESP-IDF and will propagate to Arduino next time library update happens.
The 1st point must still be added to Arduino build script, i think.

orcema, 2017-08-22:
is there a binary for windows 10 including these updates?

freeck, 2017-09-12:
After some tinkering I have arrived at the point of using source level debugging (jtag & openocd) with Eclipse. It seems that one issue remains i.e. issue 3. Has this been fixed in the latest release (I didn't saw it listed in the release notes)?

freeck, 2017-09-14:
I solved this problem in Eclipse, by:

1: Setting the parameters in Eclipse->DebugConfiguration->GdbHardwareDebuging->Startup as follows:
monitor reset halt
x $a1=0
thb loop
thr 0   // this was my lucky guess, application should be started in the APP-CPU.
c

2: Include file _FreeRTOS-openocd.c_ the project...

Using gdb also works: >_xtensa-esp32-elf-gdb.exe -x application.elf_ however an error is generated:
_esp32: target state: halted
0x0:    0x00000000
Thread ID 0 not known._
Type ctrl-C and you are able to set breakpoints , single step etc....

closed 2018-03-06 by FayeY