espressif/openocd-esp32 #14: Problem with libwebsocket application

johsely1234, 2017-08-08:
I am trying to debug and/or flash the esp32 module with openOcd. I am using a program that uses libwebsockets.
https://github.com/warmcat/lws-esp32-factory

With the example program "hello world" from esp-idf the debugger works (see below) but with the libwebsockets template app it does not work.

Error: JTAG scan chain interrogation failed: all ones

I know this problem normally occurs if the hardware connections are not right, but they are ok because the standard program is working.

Libwebsockets application(not working):
```
bin/openocd -s share/openocd/scripts -f interface/ftdi/olimex-jtag-tiny.cfg -f board/esp-wroom-32.cfg 
Open On-Chip Debugger 0.10.0-dev-ga859564 (2017-07-24-16:16)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : esp32: Debug controller was reset (pwrstat=0xFF, after clear 0xFF).
Info : esp32: Core was reset (pwrstat=0xFF, after clear 0xFF).
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: Exception reading pc!

Info : Target halted. PRO_CPU: PC=0x00000000             APP_CPU: PC=0x00000000 (active)
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32: Failed writing 4 bytes at address 0x3FF5F064, data - a1, 3a, d8, 50, a1, 3a, d8, 50
embedded:startup.tcl:21: Error: 
in procedure 'init' 
in procedure 'ocd_bouncer' 
in procedure 'flash' 
in procedure 'ocd_bouncer' 
in procedure 'esp_core_halt' 
in procedure 'esp32' called at file "share/openocd/scripts/target/esp32.cfg", line 46
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21
```

Hello world application(working):
```
bin/openocd -s share/openocd/scripts -f interface/ftdi/olimex-jtag-tiny.cfg -f board/esp-wroom-32.cfg 
Open On-Chip Debugger 0.10.0-dev-ga859564 (2017-07-24-16:16)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x40080F88 (active)    APP_CPU: PC=0x00000000
```

Setting the jtag clock speed to 1000kHz does not help.
The serial output just shows the normal program output.

The size of the bootloader binaries is different:
hello world: 15,8 kB
lws application: 11,7kB

The lws application uses a custom partition table. See attached.

I really have no clue what the problem could be here. Maybe it is also a problem from libwebsockets. 

[image]

krzychb, 2017-08-08:
The errors you report look similar to what is documented in paragraph [Do not use JTAG pins for something else](http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-jtag-pins-reconfigured). This suggest issue with JTAG pins being also used by lws-esp32-factory.

Apparently lws-esp32-factory is using GPIO14 - see [Optional default peripherals](https://github.com/warmcat/lws-esp32-factory#optional-default-peripherals).

If not done already I would change this pin to some other not used by JTAG.

lws-team, 2017-08-08:
Lws per se won't intentionally mess with your jtag; it doesn't care about it.

It sounds as @krzychb helpfully suggests it's IO or something in your config.

Again LWS itself doesn't care about your gpio, but the test apps are built around a button and LED, one gpio each.  I tried to make them easily configurable but the IO mux in ESP32 didn't make that reasonable to do at least at the time I tried to do it.

johsely1234, 2017-08-08:
Ok thanks for the fast reply. I did not think about the gpios...

In lws-plat-esp32.c there is the gpio 14 used. GPIO_SW is defined as 14 -> I changed it to 16 but that did not help unfortunately.

I also uncommented this part:

```
//	ledc_timer_config_t ledc_timer = {
//	        .bit_num = LEDC_TIMER_13_BIT,
//	        .freq_hz = 5000,
//	        .speed_mode = LEDC_HIGH_SPEED_MODE,
//	        .timer_num = LEDC_TIMER_0
//	};
//	int n;
//
//	ledc_timer_config(&ledc_timer);
//
//	lws_set_genled(LWSESP32_GENLED__INIT);
//
//	/* user code needs to provide lws_esp32_leds_timer_cb */
//
//        leds_timer = xTimerCreate("lws_leds", pdMS_TO_TICKS(25), 1, NULL,
//                          (TimerCallbackFunction_t)lws_esp32_leds_timer_cb);
//        scan_timer = xTimerCreate("lws_scan", pdMS_TO_TICKS(10000), 0, NULL,
//                          (TimerCallbackFunction_t)lws_esp32_scan_timer_cb);
//        debounce_timer = xTimerCreate("lws_db", pdMS_TO_TICKS(100), 0, NULL,
//                          (TimerCallbackFunction_t)lws_esp32_debounce_timer_cb);
//
//#if !defined(CONFIG_LWS_IS_FACTORY_APPLICATION)
//        mdns_timer = xTimerCreate("lws_mdns", pdMS_TO_TICKS(5000), 0, NULL,
//                          (TimerCallbackFunction_t)lws_esp32_mdns_timer_cb);
//#endif
//	scan_timer_exists = 1;
//        xTimerStart(leds_timer, 0);
//
//	*(volatile uint32_t *)PERIPHS_IO_MUX_MTMS_U = FUNC_MTMS_GPIO14;
//
//	gpio_output_set(0, 0, 0, (1 << GPIO_SW));
//
//	n = gpio_install_isr_service(0);
//	if (!n) {
//		gpio_config_t c;
//
//		c.intr_type = GPIO_INTR_NEGEDGE;
//		c.mode = GPIO_MODE_INPUT;
//		c.pin_bit_mask = 1 << GPIO_SW;
//		c.pull_down_en = 0;
//		c.pull_up_en = 0;
//		gpio_config(&c);
//
//		if (gpio_isr_handler_add(GPIO_SW, gpio_irq, NULL))
//			lwsl_notice("isr handler add for 14 failed\n");
//	} else
//		lwsl_notice("failed to install gpio isr service: %d\n", n);
```

I am not looking into my code if there is a gpio used...

johsely1234, 2017-08-08:
I made a endless loop at the beginning of the app_main(). This should ensure that no pin is used?

Still the same error.

krzychb, 2017-08-08:
I would not modify the source code but change the "GPIO for Force Factory Boot" in menuconfig instead.

krzychb, 2017-08-08:
> I made a endless loop at the beginning of the app_main(). This should ensure that no pin is used?

I do not think this will help.
This GPIO is configured by [bootloader code](https://github.com/warmcat/lws-esp32-factory/blob/a3d0f58d4f86202c05e6155cdaa83866ae326d60/components/bootloader/src/main/bootloader_start.c#L238), before the application gets to the `app_main()`.

johsely1234, 2017-08-08:
Yes you are right, I have to change it in the sdkconfig/make menuconfig.

But I can't find it in the make menuconfig (It should be under Security features, but I could not find it anywhere else).
It is in the sdkconfig file, but this should not be edited and does not work to change anyway.

I use esp-idf v2.1-rc1.
I can't use the latest libwebsocket because of:

```
/home/esp32dev/esp/lws-esp32-factory-master2# git clone https://github.com/warmcat/lws-esp32-factory.git
Cloning into 'lws-esp32-factory'...
remote: Counting objects: 516, done.
remote: Total 516 (delta 0), reused 0 (delta 0), pack-reused 516
Receiving objects: 100% (516/516), 870.51 KiB | 511.00 KiB/s, done.
Resolving deltas: 100% (149/149), done.
Checking connectivity... done.
root@esp32dev-VirtualBox:/home/esp32dev/esp/lws-esp32-factory-master2# cd lws-esp32-factory/
root@esp32dev-VirtualBox:/home/esp32dev/esp/lws-esp32-factory-master2/lws-esp32-factory# git submodule update --init
Submodule 'components/libwebsockets' (https://github.com/warmcat/libwebsockets) registered for path 'components/libwebsockets'
Cloning into 'components/libwebsockets'...
remote: Counting objects: 15470, done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 15470 (delta 9), reused 13 (delta 6), pack-reused 15450
Receiving objects: 100% (15470/15470), 12.45 MiB | 541.00 KiB/s, done.
Resolving deltas: 100% (11774/11774), done.
Checking connectivity... done.
fatal: reference is not a tree: 260be349a0aad3a913271aa04db92315899659cc
Unable to checkout '260be349a0aad3a913271aa04db92315899659cc' in submodule path 'components/libwebsockets'
```

johsely1234, 2017-08-08:
Ok it was the pin. I could change it in the file components/bootloader/src/Kconfig.projbuild in my application directory.

config BOOTLOADER_FACTORY_BUTTON_GPIO
	int "GPIO for Force Factory Boot"
	default 16

I don't know why it is not shown in menuconfig, but that would be better to change it there.

Thanks a lot for the help!

krzychb, 2017-08-08:
We are getting to the point of learning internals of this application so, JTAG debugging will not be required :smile: 

The option is under : `Security features > Potentially insecure options  >  (14) GPIO for Force Factory Boot` 

[image]

johsely1234, 2017-08-08:
Don't worry I will need the debugger for the application for sure ;-)

Yeah I found it now, I had to activate "Enable secure boot in bootloader" or "Enable flash encryption on boot". 
I don't need this two options for now. I should be able to set these if both options are disabled or not?

johsely1234, 2017-08-08:
It should be in the Component config -> Libwebsockets right?

lws-team, 2017-08-09:
When I did it, I recall it appearing somewhere more useful... maybe it also depends on esp-idf.  If there's something it should declare better patches are very welcome.

"the button" is used to force us to come up in factory, to unfortunately it has to get checked in the bootloader.

closed 2017-08-10 by johsely1234

zhivko, 2018-11-15:
No (14) GPIO for Force Factory Boot option in esp-idf esp-idf-v3.1.1.
I am using olimex adapter arm-usb-ocd-h and could not debug esp32 wroom by following instructions in esp-idf.

I tried with connecting VDD on olimey 20pin jtag adapter and esp32 3.3V pin.
[image]

`
bin/openocd -s share/openocd/scripts -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: level must be between -3 and 3
debug_level number

adapter speed: 20000 kHz
esp32 interrupt mask on
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors

Info : accepting 'gdb' connection on tcp/3333
Error: Target not examined yet

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
`

d3 log from openocd attached
[olimex_d3.txt](https://github.com/espressif/openocd-esp32/files/2583168/olimex_d3.txt)