espressif/openocd-esp32 #15: start the commands xtensa-esp32-elf-gdb -x gdbinit build/blink.elf and the openocd reports the error below

rechan886, 2017-08-08:
Error: cpu0: esp32_fetch_all_regs (line 234): DSR (8011CC13) indicates DIR instruction generated an exception!
Info : Target halted. PRO_CPU: PC=0x400D16B3 (active)    APP_CPU: PC=0x4000BFEA 
Error: FreeRTOS maximum used priority is unreasonably big, not proceeding: 1061164148

and I can not debug the program

zjykymt, 2017-08-11:
I also had the same problem.

Include two log files
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/1217293/gdb_log.txt)
[openocd.txt](https://github.com/espressif/openocd-esp32/files/1217297/openocd.txt)

zjykymt, 2017-08-11:
@rechan886 There is something worth sharing is that I have just succeeded in debugging.

The solution is that I am modifying "adapter_khz" parameter within "esp-wroom-32.cfg":
old:adapter_khz 20000 
new:adapter_khz 2000

JTAG Wiring Table(JLINK same applies)
[image]

krzychb, 2017-08-11:
@zjykymt, thank you for sharing your findings!
How long is the cable you are using to connect ESP32 to JLINK? Could you post a picture?

rechan886, 2017-08-12:
@zjykymt thank you for your suggestion. The problem has solved.

zjykymt, 2017-08-15:
@krzychb The cable length is 20cm, for the 20M frequency may be too long.

krzychb, 2017-08-16:
> The wire length is 20cm(every wire equal length), for the 20M frequency may be too long.

@zjykymt, I think this is the case. I cut "standard" DuPont breadboard 20 cm wires into halves. 
With 10 cm equal length wires I could run JTAG @ 20 and 26 MHz without any issues.

krzychb, 2017-08-24:
> @zjykymt thank you for your suggestion. The problem has solved.

@rechan886, could you close this issue report?

closed 2017-08-24 by rechan886