espressif/openocd-esp32 #16: Issue when running Espruino

chevdor, 2017-08-25:
I am using a ESP32-WROOM-32 and an Olimex ARM-USB-OCD-H.

I tried connecting openocd while the ESP32 runs a blink program compiled with the Espressif SDK and everything works as expected. I can for instance connect and halt the CPU.

When running Espruino on the same board and the same connections however, things don´t go so smooth:
```
$ openocd -s tcl -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev-ged7b1a9f (2017-07-11-15:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : esp32: Debug controller was reset (pwrstat=0xFF, after clear 0xFF).
Info : esp32: Core was reset (pwrstat=0xFF, after clear 0xFF).
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: Exception reading pc!

Info : Target halted. PRO_CPU: PC=0x00000000             APP_CPU: PC=0x00000000 (active)
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32: Failed writing 4 bytes at address 0x3FF5F064, data - a1, 3a, d8, 50, a1, 3a, d8, 50
embedded:startup.tcl:21: Error: 
in procedure 'init' 
in procedure 'ocd_bouncer' 
in procedure 'flash' 
in procedure 'ocd_bouncer' 
in procedure 'esp_core_halt' 
in procedure 'esp32' called at file "tcl/target/esp32.cfg", line 37
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21
```

I can connect with telnet/4444 but this time, commands such as `halt` have simply no effect at all.

krzychb, 2017-08-27:
Hi @chevdor,

Above error log resembles log described in http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/configure-other-jtag.html#configure-hardware

Checking the picture under "GPIO Pins" on page  http://www.espruino.com/ESP32 it looks to me that Espruino may be configuring all [ESP32 JTAG pins](http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/configure-other-jtag.html#configure-hardware) (GPIO12 - 15) as ADC pins / inputs. If this is the case then JTAG will not work.

If not done already I would ask Espruino team if this is the case.

closed 2017-11-23 by igrr

SunilDullolli, 2018-08-28:
@chevdor, What did you do to get over this problem? I have the same problem, the JTAG works fine when I program the ESP32_EVB launchpad but I get this error when I try to program brand new ESP32 which is present on one of our PCB's. I am pretty sure that JTAG pins are not used for something else. Any suggestions are appreciated. Thanks