espressif/openocd-esp32 #17: Error: buspirate debug interface not found

DavidAntliff, 2017-09-04:
I get an error when trying to use a BusPirate as my JTAG debug interface.

With the original v0.10.0 from openocd.org:

```
$ openocd -f buspirate.cfg 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'buspirate' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
Info : Buspirate Interface ready!
Info : This adapter doesn't support configurable speed
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
Info : TAP auto0.tap does not have IDCODE
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 2 -expected-id 0x00000000"
Error: auto0.tap: IR capture error; saw 0x0002 not 0x0001
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
```

But with this fork, built from cc5b0f7a0215e5c121f63d9c884176ca9627d9d2:

```
$ src/openocd -f buspirate.cfg 
Open On-Chip Debugger 0.10.0-dev-gcc5b0f7a (2017-09-04-13:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (buspirate)
The following debug interfaces are available:
1: ftdi
2: usb_blaster
3: usbprog
4: jlink
5: vsllink
6: rlink
7: ulink
8: arm-jtag-ew
9: hla
10: osbdm
11: opendous
12: aice
13: cmsis-dap
```

My `buspirate.cfg` looks like this:

```
interface buspirate
buspirate_port /dev/cu.usbserial-A7004q3B
buspirate_speed normal
buspirate_vreg 0
buspirate_mode open-drain
buspirate_pullup 1
reset_config srst_only
```

Is the `buspirate` debug interface explicitly disabled in this fork? If so, can you tell me why please?

DavidAntliff, 2017-09-04:
I discovered that no changes were made in this fork to disable this debug interface. Instead, `configure` has an `--enable-buspirate` option that enables this functionality. I will therefore close this issue as it is invalid.

closed 2017-09-04 by DavidAntliff

ThomasBarth, 2019-05-28:
Could you please provide a Windows build with BusPirate support

gerekon, 2022-10-25:
That interface is not supported on MinGW. It can be build for cygwin host only. Our releases are build for MinGW only. Support for Cygwin host build is not planned yet.