espressif/openocd-esp32 #18: Setup of OpenOCD instructions incomplete for OS X

csann, 2017-09-11:
On this page 

[](https://dl.espressif.com/doc/esp-idf/latest/api-guides/jtag-debugging/setup-openocd-macos.html)

the instructions are incomplete. In addition to downloading and extracting the file to the ~/esp folder, you must also install libusb as follows:

a.  Install Homebrew by typing in Terminal: "ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)" < /dev/null 2> /dev/null"
and  b. Install libusb by typing in Terminal: "brew install libusb" 

If you add these instructions to your web page you will save your OS X users hours of wasted time and frustration.

Once I confirm a few things, I will probably have further suggestions for improving your instructions. Is this the proper method to submit this type of suggestions?

Thanks!

igrr, 2017-09-12:
Thank you for reporting this. If you wish to submit a PR, you can update this file: https://github.com/espressif/esp-idf/blob/master/docs/api-guides/jtag-debugging/setup-openocd-macos.rst

csann, 2017-09-12:
I’m not too good at using git. I’ll let you guys fix it.

closed 2018-05-25 by igrr