espressif/openocd-esp32 #19: Errors building on Mac OS with Xcode 9

pieceofsummer, 2017-09-19:
One in esp108.c:
```bash
esp108.c:1312:6: error: logical not is only applied to the left hand side of this bitwise operator [-Werror,-Wlogical-not-parentheses]
        if (!intfromchars(traxctl)&TRAXCTRL_TREN) {
            ^                     ~
esp108.c:1312:6: note: add parentheses after the '!' to evaluate the bitwise operator first
        if (!intfromchars(traxctl)&TRAXCTRL_TREN) {
            ^
             (                                  )
esp108.c:1312:6: note: add parentheses around left hand side expression to silence this warning
        if (!intfromchars(traxctl)&TRAXCTRL_TREN) {
            ^
            (                     )
1 error generated.

```

Another one in esp32.c:
```bash
esp32.c:1611:6: error: logical not is only applied to the left hand side of this bitwise operator [-Werror,-Wlogical-not-parentheses]
        if (!intfromchars(traxctl)&TRAXCTRL_TREN) {
            ^                     ~
esp32.c:1611:6: note: add parentheses after the '!' to evaluate the bitwise operator first
        if (!intfromchars(traxctl)&TRAXCTRL_TREN) {
            ^
             (                                  )
esp32.c:1611:6: note: add parentheses around left hand side expression to silence this warning
        if (!intfromchars(traxctl)&TRAXCTRL_TREN) {
            ^
            (                     )
1 error generated.
```

amcjen, 2017-11-13:
Seconding this bug.  I also experience it on OS X 10.12.6

igrr, 2017-11-23:
Resolved in befba2b

closed 2017-11-23 by igrr