espressif/openocd-esp32 #20: registerview in eclipse debugger

igrr, 2017-09-23:
_From @hackrid on September 16, 2017 18:5_

Hi,

hows your thinking regarding adding a register description file to embsysregview for esp8266 and esp32?
would improve the debugging expirience a lot. 

http://embsysregview.sourceforge.net/

best regards

_Copied from original issue: espressif/esp-idf#1007_

igrr, 2017-09-23:
Thanks for the suggestion @hackrid, we will prepare the register description file.
I have moved the issue to openocd repository as it is more related to debugging features such as this one.

hackrid, 2017-09-23:
thanks very much for picking this up! I appreciate it very much! keep us posted :)

manoj153, 2019-05-13:
any update on when it's expected to release? I am finding ways how to debug esp32 with register and memory view

gerekon, 2022-09-01:
Close this issue because it is unrelated to OpenOCD. Console GDB have all necessary means to check memory and registers. If you want to have it in IDE, please check ESP-IDF [VsCode extension](https://github.com/espressif/vscode-esp-idf-extension) (implements requested features) and (Eclipse plugin)[https://github.com/espressif/idf-eclipse-plugin]

closed 2022-09-01 by gerekon