espressif/openocd-esp32 #21: xtensa-esp32-elf-gdb.exe stops working on windows 10 eclipse debugging

sharif351, 2017-10-27:
Hi,

I have been trying to step through the "https_request" project following the guideline of eclipse GDB debugging configuration guideline. While doing so I, after hitting a break point, on resume, code breaks at "xQueueGenericReceive() func", event though i haven't set any break point there. After that, once I hit resume button/step_over button, xtensa-esp32-elf-gdb.exe stops working, and my debugging session stops. Please see the attached files for all the OCD and GDB logs. Thnaks!

[image]

[image]

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/1422470/openocd_log.txt)

[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/1422473/gdb_log.txt)

Regards,
Sharif

BobaTrek, 2017-12-14:
I am having the exact same issue.  Did you ever get this solved?

chegewara, 2017-12-19:
I would start from here:
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/March_2008/freertos_Queue_Problem_1971146.html

sharif351, 2017-12-19:
Hi,
My apology for my delayed response! Anyway, I got this issue solved for my case. I was using WROVER kit and I had the following pins (please see in the following picture) sorted; which I was not supposed to do. After removing the short it started working fine. 
Thanks!
[image]

BobaTrek, 2017-12-19:
You are a life saver!

I am evaluating if my company wants to base product on the ESP32, and this would have been a show stopper.

In case anybody else finds this thread by searching:

The problem for me was that the documentation for the ESP WROVER KIT here is incorrect:
http://esp-idf.readthedocs.io/en/latest/get-started/get-started-wrover-kit.html
On this page, in "Setup Options" for the JTAG jumper JP8 it says to install 5 jumpers to enable JTAG. However, apparently the 5th jumper labelled S_TDI should not be installed.  Only 4 jumpers should be installed as depicted in @sharif351 's attachment.

I do still occasionally hit breakpoints (that I did not set) in queue.c, but the debugger no longer crashes at that point and I can usually step my way back into the thread I was trying to step through.

closed 2021-01-11 by gerekon