espressif/openocd-esp32 #22: Custom SystemView events not supported

chrta, 2017-10-30:
I would like to add my own event to my SystemView trace. In my firmware i added this besides enabling the SystemView tracing functionality:
~~~
SEGGER_SYSVIEW_RecordU32(apiSlowID_OFFSET + apiID_MY_EVENT_INDEX, payload)
~~~

But when i start the trace vie openocd, i get:
~~~
SEGGER: Unsupported event ID 383!
~~~

This is generated in this line: https://github.com/espressif/openocd-esp32/blob/master/src/target/esp108_apptrace.c#L1797

Is there a generic way to enable custom events without the need for everyone to modify the openocd source and recompile openocd?
Is it possible to just remove the check `(event_id > SYSVIEW_EVENT_ID_MAX)`?

chrta, 2017-11-02:
FYI: I removed to following four lines from the source code and it worked for me.

~~~
if (event_id > SYSVIEW_EVENT_ID_MAX) {
  LOG_ERROR("SEGGER: Unsupported event ID %d!", event_id);
  return res;
}
~~~

igrr, 2018-11-05:
Closing as fixed in e3029dad40c7e2704bb4e0004e92b8edb8f593b7.

closed 2018-11-05 by igrr