espressif/openocd-esp32 #23: Upstream?

ligius-, 2017-11-01:
Will the changes ever be pushed to the upstream repository? Or will this fork keep up with the main one? There are quite a few new fixes and features upstream and I found myself having to manually integrate this fork as there were some breaking changes upstream (for example src/target/Makefile.am)

cz172638, 2017-11-23:
rebase on top of openocd master and submit for review to openocd gerrit.

satoshinm, 2017-12-23:
Hi, I've started an attempt to reconcile the merge conflicts with openocd-esp32 and the current upstream openocd as part of upstreaming it, progress here: https://github.com/satoshinm/openocd/pull/2 - so far it compiles but I have no way yet to test it, maybe helpful as a starting point?

brainstorm, 2021-01-30:
Here's a DirtyJTAG merge (forks ntfreak+espressif+dirtyjtag repos in one)... if somebody is interested in testing DirtyJTAG, which can be flashed on sub-dollar STLinkv2 clones.

Admittedly it's not a very clean merge, I just wanted to get ESP32+DirtyJTAG working, but perhaps can be used to cherrypick and (finally?) upstream this fork at some point in the future?

More context on issue https://github.com/jeanthom/openocd-dirtyjtag/issues/3.

cederom, 2021-11-07:
Any updates and / or plans on share back to the upstream?

gerekon, 2021-11-08:
Hi @cederom 

We plan to start upstreamimg our changes the next year. I hope in the beginning of it. Now we preparing the merge with the recent OpenOCD upstream changes and going onto v0.11

gerekon, 2022-10-25:
This year we started upstreaming our changes. The work is in progress. Some chips are already supported. Not all features yet.
https://review.openocd.org/c/openocd/+/6889  - semihosting
https://review.openocd.org/c/openocd/+/6991  - target/workarea config
https://review.openocd.org/c/openocd/+/6989  - New chip support ESP32
https://review.openocd.org/c/openocd/+/6990  - New chip support ESP32-S3
https://review.openocd.org/c/openocd/+/7163  - application tracing functionality over JTAG

closed 2022-10-25 by gerekon