espressif/openocd-esp32 #27: Debug fails / stops in taskENTER_CRITICAL

nkolban, 2018-01-22:
I am running a WROVER DevKit with the latest ESP-IDF master as of this date.   When I run gdb against my ESP32, all works as desired ... for a period of time.   As I step through the statements using `n` I am presented with the next statement as desired.  However, after stepping over a varying number of statements, when I next execute `n` I do NOT wake up at the next statement, instead I wake up here:

```
108		ESP_LOGD(LOG_TAG, "Setting up callbacks for logging");
(gdb) n
Target halted. PRO_CPU: PC=0x4008256C (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x4008774D (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x4008774F (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x40087752 (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x40087755 (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x40087758 (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x4008775A (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x4008775C (active)    APP_CPU: PC=0xF8001080 
Target halted. PRO_CPU: PC=0x40083E63 (active)    APP_CPU: PC=0xF8001080 

Program received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073412716]
xQueueGenericReceive (xQueue=0x3ffaea14, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0)
    at /home/kolban/particle/Phase2/esp-idf/components/freertos/./queue.c:1455
1455			taskENTER_CRITICAL(&pxQueue->mux);
(gdb) q
```

This log shows that at line `108` of my source code, I executed `n` and then wake up at line `1455` of `freertos/queue.c:1455`.  I can not continue from here.

I can recreate this using either command line `xtensa-esp32-elf-gdb` or Eclipse debugging.

My `gdbinit` looks as follows:

```
set remotelogfile gdb_log.txt
target remote :3333
mon reset halt
thb app_main
x $a1=0
c
```

I can recreate at will using a variety of distinct applications.  My platform is Ubuntu 17.10 running under Oracle Virtual Box. 

I am attaching both:

* `gdb_log.txt`
* `openocd.log`

as requested in the reporting information.  I am available to demonstrate live or provide additional documentation on request.

[bug.zip](https://github.com/espressif/openocd-esp32/files/1650933/bug.zip)

BobaTrek, 2018-01-22:
I have experienced the same issue.  Please see my submission:
[https://github.com/nkolban/esp32-snippets/issues/296](https://github.com/nkolban/esp32-snippets/issues/296)

Another programmer also experienced this issue here:
[https://github.com/espressif/openocd-esp32/issues/21](https://github.com/espressif/openocd-esp32/issues/21)

At the end of the above two threads there is a 5 step process that shows how to partially work around the problem of unexpectedly breaking in queue.c, but it does not always work (and is very tedious)

I am really hoping this issue gets figured out!

worker2345234, 2018-02-06:
Hello,

same here! I'm using the current IDF from today and the blink example.
$ bin/openocd --version
Open On-Chip Debugger 0.10.0-dev-ga859564 (2017-07-24-16:18)

worker2345234, 2018-02-06:
Sorry, I forgot to mention information about my jtag adapter: I'm using a USB powered WROVER-Devkit V2.
On JP8 I've set 5 jumpers: TMS, TDO, TDI, TCK and S_TDI are close.
I left nSRST and GND open.

foxlav, 2018-02-07:
Hello, 
I am using a USB powered Espressif ESP32-WROVER-KIT V3 and have the same issue.

Sermus, 2018-03-03:
+1
Reproduces for me both with WROVER KIT and in house developed hardware.

igrr, 2018-03-03:
@Sermus could you please mention OpenOCD version you are using?

Sermus, 2018-03-03:
Sure.
Open On-Chip Debugger 0.10.0-dev-ga859564 (2017-07-25-00:19)

igrr, 2018-03-03:
2017-07-25 seems to be pretty dated, could you try the git head?

BobaTrek, 2018-03-03:
I am having the same issue with breakpoints.
I just checked my version and it is the same as @worker2345234:
Open On-Chip Debugger 0.10.0-dev-ga859564 (2017-07-24-16:18)

I got this version by following Espressif's esp-idf installation instructions here:
[http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/setup-openocd-windows.html](url)

Espressif's installation instructions say to download the openocd-esp32 version from this link:
[https://dl.espressif.com/dl/openocd-esp32-win32-a859564.zip](url)

Checking the GitHub for openocd-esp32 I do see numerous changes since the last time Espressif packaged a binary version for their esp32-idf toolkit and installation instructions.

Does anybody know where we can download a more recently built package of openocd-esp32?  I am very hesitant to download the source from GitHub and try to build it myself.

igrr, 2018-03-04:
When this (sigtrap in multi_heap) happens, do you see any output from the exception handler on the UART console?

Sermus, 2018-03-04:
Yes, this is the message:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
 at pc=4013ff40. Setting bp and returning..

But it never happens if OCD is not run.

igrr, 2018-03-04:
It's interesting that the PC reported by the exception handler does not match the backtrace you get from GDB. Can you see what is the output of `l *0x4013ff40` in GDB (with the same elf file that produced the exception)?

igrr, 2018-03-04:
Also, does this happen when you are running (`continue` or `next`) or when you are single stepping (`step`)?

Sermus, 2018-03-04:
Weird things happen. I can't repro sigtrap in the place i reported initially. Now it halts at another location in my user code. It crashes because it tries to dereference nullptr coming as a parameter to a function. The weird is that the caller definitely passes not null as this parameter. I.e. in the function it crashes i see the parameter is null. If i move a frame back down by the stack i see the parameter passed is not null. Stack corruption or something similar happens.

I now reproduced this twice with exactly same symptoms. If i run the app without OCD it executes normally.

I used to use only next and continue while reproducing this.

igrr, 2018-03-04:
One possibility i can think of is that some register gets zeroed out when OpenOCD is saving/restoring the context. If you can run through this repro process, can you please also check which register is holding the argument? If this is the same register every time, then it's quite probable that OpenOCD is responsible.

Sermus, 2018-03-04:
The snippet at the caller is:
401415cc:	41c8      	l32i.n	a12, a1, 16
401415ce:	b1b8      	l32i.n	a11, a1, 44
401415d0:	01ad      	mov.n	a10, a1
401415d2:	0e2132        	l32i	a3, a1, 56
401415d5:	0003e0        	callx8	a3

nkolban, 2018-03-05:
I am able to recreate as needed.  Would be delighted to offer up a screen share session if that would be of assistance.  Was the original ZIP file supplied in the first post usable in any fashion?

igrr, 2018-03-05:
@nkolban could you please check whether you are using latest OpenOCD version or the older `0.10.0-dev-ga859564 (2017-07-25-00:19)` one?

igrr, 2018-03-05:
@BobaTrek @nkolban 

By the way, the most recent builds of OpenOCD can be downloaded from the following URLs:

https://dl.espressif.com/dl/openocd-esp32-win32-07baed8c.zip
https://dl.espressif.com/dl/openocd-esp32-macos-07baed8.tar.gz
https://dl.espressif.com/dl/openocd-esp32-linux64-07baed8.tar.gz

igrr, 2018-03-05:
Ok, 2nd argument maps to a3 within the function, and a3 is used as a scratch register, so my hypothesis checks out. We'll try to reproduce this and figure out why the register is not restored.

Sermus, 2018-03-06:
@igrr Sorry for the delay. The log is attached. As far as i can see, there are several write_dirty_register segments.
[openocd_log.tar.gz](https://github.com/espressif/openocd-esp32/files/1785729/openocd_log.tar.gz)

igrr, 2018-03-06:
Ok, so right before the last `continue` there is

```
cpu1: Writing back reg a3 value 00000000, num =4
```

which means that it is being written back, but the value is possibly corrupted. Thank you for the logs, at least we now know what to look for.

nkolban, 2018-03-10:
Following the suggestion from @igrr to install https://dl.espressif.com/dl/openocd-esp32-linux64-07baed8.tar.gz ...  I have done this and now seem to be resolved relative to the original content of this issue.  If it were just my issue, I'd close it with thanks ... but I get the sense that there may now be others leveraging the same issue for either related or distinct content.  Others ... please review and determine if your issues are the same as this one and determine if a separate issue is warranted.   From my perspective, as the originator of the issue ... I am not satisfied and happy.

worker2345234, 2018-03-12:
Hi, with the openocd version from https://dl.espressif.com/dl/openocd-esp32-win32-07baed8c.zip, the debugging works for me. Thanks!

worker2345234, 2018-03-12:
@igrr Please update the programming guide, e.g. https://dl.espressif.com/doc/esp-idf/latest/api-guides/jtag-debugging/setup-openocd-windows.html

krzychb, 2018-03-12:
@worker2345234, do you mean to provide the links to the most recent builds of the OpenOCD?

worker2345234, 2018-03-12:
yes, because the mentioned https://dl.espressif.com/dl/openocd-esp32-win32-a859564.zip obviously does not work at all

igrr, 2018-04-18:
@Sermus The issue about `a3` corruption has been fixed, please try latest version.

closed 2018-04-18 by igrr

worker2345234, 2018-04-18:
@igrr Do you have a link for the latest version (the windows binary)?

igrr, 2018-04-18:
https://github.com/espressif/openocd-esp32/releases