espressif/openocd-esp32 #29: Target Halted SIGTRAP - Keeps interrupting debug session

neilpanchal, 2018-02-18:
I have noticed (even in the esp32 docs regarding gdb) that there are tons of hardware level halts causing the debugging session to be completely counter productive vs. just using print statements.

I have since added the following to my gdbinit file:

```
handle SIGTRAP noprint nostop pass
target remote :3333
mon reset halt
thb app_main
x $a1=0
c
```
I got the SIGTRAP interruptions from stopping everything but I still get loads and loads of print outs in the terminal:

```
Target halted. PRO_CPU: PC=0x400855DF (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400855E2 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400855E5 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400855E7 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400855E9 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x4000BFF0 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x4000BFF3 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x4000BFF6 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x40085718 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x40081391 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1AA8 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1AD6 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1AAF (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1AB2 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1AB5 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1AB7 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1ABA (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x400D1ABC (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x40084C54 (active)    APP_CPU: PC=0x400D2C36
Target halted. PRO_CPU: PC=0x40084C57 (active)    APP_CPU: PC=0x400D2C36
|
|
|
v
Infinity (wait 30 seconds for all the print outs to complete and then I am greeted with the gdb line I was on
```
How can we make the debugging session **actually** useful?

igrr, 2018-03-16:
Hi @neilpanchal, this looks quite similar to #27. Could you please give the latest OpenOCD version a try? You can find download links in this comment: https://github.com/espressif/openocd-esp32/issues/27#issuecomment-370319753.

BobaTrek, 2018-03-16:
+1 on this issue

I too had the problem #27.  I downloaded the new openocd zip file posted in #27 and the problem of the debugger frequently stopping in queue.c did go away.  Yeah!

This is now the version of openocd I am running on a WROVER KIT:
$ bin/openocd --version
Open On-Chip Debugger 0.10.0-dev-g07baed8c (2018-02-23-08:58)

However, even with the new version of openocd, the problem described here (tons of Target halted prints) does occur on my setup.

These numerous prints can make each single singlestep command take many seconds as the prints continuously scroll on the screen.  The prints will finally stop, and then the debugger in Eclipse appears on the next line of code.

igrr, 2018-03-16:
Okay, so basically we need to make these steps run faster, right?

BobaTrek, 2018-03-16:
Single stepping faster would be a great benefit.  I am not sure if it is the prints that are clogging the system or something else and the prints are an artifact.

When single stepping, the type of line of code seems to affect how many prints show up in the openocd-esp32 console and consequently how long you then have to wait for them to finish before Eclipse comes back to life at the next line of code.

One line I found that prints a lot and takes longer than most is single stepping over this line:
std::stringstream strm;

Hope this info helps!

erhankur, 2023-01-17:
Closing due to inactivity.

closed 2023-01-17 by erhankur