espressif/openocd-esp32 #35: Receiving JTAG scan chain interrogation failed: all ones

Gnorkus, 2018-04-04:
Hi,

Gnorkus, 2018-04-04:
I'm going to try Ubuntu 17 as a VirtualBox, but I'm not sure if there will be any difference.

Gnorkus, 2018-04-05:
Same problem in Ubuntu 17.10.  I also tried on a different Windows 10 machine, ASUS brand.  Better specs than my Surface Pro 2017 I7.  I have a list of what I've done for the setup, and the results are identical.  Fine for the build.  Just doesn't do anything.  

I have a thread on https://www.esp32.com under the title "WRover-Kit Attach to Ubuntu 16.04LTS VirtualBox".  It describes everything done to date and includes my .profile file and screenshots of the board I'm using.  There's NOTHING attached to it, and it only has the 4 JTAG jumpers (not the 5 in the picture, since that doesn't work either).  The software being run is simply the esp-idf/examples/get-started/hello_world sample code, including it's sdkcongfig file....

Thanks!

krzychb, 2018-04-05:
@Gnorkus, this is impressive. Congratulations on persistence and finding the root cause :+1:

Gnorkus, 2018-04-05:
Thanks.  If Espressif is watching this, it would be nice if they sent me 6 or so dozen WRover modules to ease the pain of having solved it.  It might also help my boss uncross his eyes.

ilxj, 2019-06-01:
Hi ，
  I got the trouble ，I use the ESP-WROVER-KIT and in windows7 when I run the cmd：
```
bin/openocd.exe  -s share/openocd/scripts -f share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f share/openocd/scripts/board/esp-wroom-32.cfg
```
and the error info ：

```
# bin/openocd.exe  -s share/openocd/scripts -f share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f share/openocd/scripts/board/esp-wroom-32.cfg
Open On-Chip Debugger  v0.10.0-esp32-20190313 (2019-03-13-09:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
```
what can I do ???

oliverjc, 2019-12-22:
Hi, I just want to add my solution to a similar problem. Setting Serial Flasher config --> SPI Flash speed to 20MHz got my JTAG working on my LyraT board. I didn't see that I had to do that anywhere in the documentation. 

It might also be worth noting that I rolled back my version of OpenOCD to `openocd-esp32-macos-0.10.0-esp32-20190708.tar`. Don't know if that makes a difference, but at this point, I'm too scared to update again 😅

arossoinsite, 2020-02-04:
> SOLVED
> It turns out I had a defective board. There was a ground short UNDERNEATH the ESP32-WRover module. How was this determined...
> 
>     1. Figured it might be a problem with the jtag pins, so ground to pin continuity was checked on the TMS TDO TDK and TCK pins of J8 (the pins closer to the inside of the board). Continuity was detected at TCK pin. 0.4 ohms.
> 
>     2. Removed R50. It was indeed a 10k 0402 resistor as per schematic. Checked both of the pads for continuity to ground, and both were grounding.
> 
>     3. Traced line back to IO13 on the WRover module. Determined that it's neighboring pin was a ground pin, so removed R37. Once removed, only one of the R50 pads was now having continuity to ground.
> 
>     4. Removed WRover Module from board using hot air rework tool. Discovered this board was actually designed to use WRoom or WRover, and that a couple of the pads for the WRoom were shorted by a solder blob. Those pads connected to ground and IO13 on the WRover module.
> 
>     5. Soldered on new WRover module.  Tested flashing and OpenOCD operation.   It now works.  See post under the title "WRover-Kit Attach to Ubuntu 156.04LTS" for picture of correct operation.
> 
>     6. Will never take for granted that the factory did their QA inspection.

I have the same problem.
Can you tell me with more details how you managed to remove the WRover module from the board?
Thanks!

v1993, 2020-06-13:
I had the same error message and found a solution: I had to connect USB cable from PC to ESP32 as well as power pins from adapter. While ESP32 appeared to be working fine without USB power, JTAG did not.

Hope this will help someone else.