espressif/openocd-esp32 #37: GDB crashes on returning to a caller function

Sermus, 2018-04-29:
Scenario:
1) Set a breakpoint somewhere
2) Do several "step over" steps
3) Do "Step in" into a callee
4) Do "step over" until execution reaches end of the function
5) Do "step over"
Expected result: execution returns to a caller function
Actual result: GDB crashes with the following traces:

/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/regcache.c:697: internal-error: enum register_status regcache_raw_read_unsigned(struct regcache *, int, ULONGEST *): Assertion `regnum >= 0 && regnum < regcache->descr->nr_raw_registers' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/regcache.c:697: internal-error: enum register_status regcache_raw_read_unsigned(struct regcache *, int, ULONGEST *): Assertion `regnum >= 0 && regnum < regcache->descr->nr_raw_registers' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

OpenOCD version: Open On-Chip Debugger 0.10.0-dev (2018-04-29-18:33) built from 039d2e86
GDB version: GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10

[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/1958891/gdb_log.txt)
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/1958893/openocd_log.txt)

ivankravets, 2018-04-29:
I also very often experience the same error with "regnum >= 0 && regnum < regcache->descr->nr_raw_registers' failed".

Sermus, 2018-06-02:
Is there any progress on this please?
This issue makes debugging near impossible in complex environments .

igrr, 2018-06-02:
The fix for this will be included into the next toolchain update, which may take a few weeks to finish. However we do have a workaround for this on OpenOCD side, which is in review at the moment. Should be merged next week.

closed 2018-06-04 by igrr

ashigupta, 2019-11-28:
Hi, is there any solution available for "JTAG debug of blink due to: Error: Too large number of threads 2044975697! #1928". 
I am facing this issue a lot and this is making debugging near to impossible for me.

rebeliousconformist, 2020-01-09:
+1  any update on the same ?

gerekon, 2020-05-04:
@ashigupta @rebeliousconformist @exislow This issue seems to be closed. Do you observe exactly the same problem as described by topic starter?