espressif/openocd-esp32 #38: Debugger halts at app_main, does not show threads.

Greg-R, 2018-05-06:
I am trying to get an ESP-Wrover-Kit working with USB JTAG.  This board has the WROVER module.

I am using ESP-IDF master branch updated as of May 4 2018.
OpenOCD:
Open On-Chip Debugger 0.10.0-dev (2018-04-18-12:03)

Build tools and GDB:
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a)

I built and flash the example "blink" program.  It works, I can see the screen blinking.
I followed the instructions for setting the board with jumpers and setting up software etc.
I am using Ubuntu Mate with kernel:
4.13.0-39-generic #44-Ubuntu SMP Thu Apr 5 14:25:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
Running on Lenovo Thinkserver.  This machine and Mate work fine with ESP32 development boards.

Log files for OpenOCD and GDB are attached.  I also attached stdout which illustrates what happens.  It does not show the "threads" and does not get to the (gdb) prompt as it shows on the web page.

Regards,
Greg
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/1978060/openocd_log.txt)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/1978061/gdb_log.txt)
[stdout_debugging_session.txt](https://github.com/espressif/openocd-esp32/files/1978062/stdout_debugging_session.txt)

igrr, 2018-05-25:
Hi @Greg-R, based on the log it seems that the breakpoint does not get hit.

Sometimes this happens if the elf file you are passing to GDB does not match the application flashed into the target. Could you please confirm if target is running the same code as you have in the elf file (i.e. by building and flashing the app again)?

igrr, 2018-05-29:
Thanks for the extra info! Could you please attach your `gdbinit` file as well?

frederic, 2018-06-05:
Hello,

I have the same issue.
I am using the default GDB file : 
```
target remote :3333
mon reset halt
thb app_main
x $a1=0
c
```
Tried with two different Linux computers, also tried both release packages or sources (esp-idf,openocd).

I would like to add that the JTAG programming also failed, despite that TAPs are detected :
```
bin/openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover.cfg -c "program_esp32 filename.bin 0x10000 verify exit"
Open On-Chip Debugger 0.10.0-dev (2018-05-31-23:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : Target halted. PRO_CPU: PC=0x400E16E2 (active)    APP_CPU: PC=0x400E16E2 
esp32: target state: halted
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
** Programming Started **
auto erase enabled
Info : Use core0 of target 'esp32'
[WAIT FEW SECONDS...]
Error: timed out while waiting for target halted / 4 - 2
Info : Target halted. PRO_CPU: PC=0x40091138 (active)    APP_CPU: PC=0x400076E0 
Error: xtensa_wait_algorithm: not halted 0, pc 0x40091138, ps 0x60021
Error: Faied to wait algorithm (0)!
Error: Algorithm run faied (-302)!
Info : Auto-detected flash size 0 KB
Error: Failed to probe flash, size 0 KB
Error: Bank is invalid
** Programming Failed **
shutdown command invoked
```

A last detail, I noticed two additional lines in both GDB & OpenOCD logs (in @Greg-R 's log as well ) : 
```
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
```

igrr, 2018-06-05:
@frederic Could you please try replacing `x $a1=0` command in gdbinit file with `set $a1=0`? Additionally, please check that application does not use GPIOs of the JTAG interface (GPIO12-15).

frederic, 2018-06-05:
Hello @igrr 

Thank you for your reply.

I am using the blink sample code from esp-idf, and the only GPIO that seems to be used is CONFIG_BLINK_GPIO=5 .

I've replaced `x $a1=0` with `set $a1=0` in gdbinit, here's the log :
```
$ xtensa-esp32-elf-gdb -x gdbinit ./build/blink.elf 
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/blink.elf...done.
0x400e16e2 in esp_vApplicationWaitiHook () at /data/user/work/esp/src/esp-idf/components/esp32/freertos_hooks.c:66
66	    asm("waiti 0");
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
Hardware assisted breakpoint 1 at 0x400d200f: file /data/user/work/esp/src/esp-idf/examples/get-started/blink/main/blink.c, line 43.
```
We can notice that a difference with the initial gdb config (`x $a1=0`) : the following line is now missing at the end of log :
```
0x0:	0x00000000
```

Thank you

frederic, 2018-06-05:
JFYI, here are the steps I am following :
1) Build & flash blink project using `make flash`. Disconnect USB, remove UART jumpers.
2) Connect to USB : LCD blinks (ESP-WROVER-KIT)
3) Run OpenOCD, LCD still blinks :
```
$ ./src/openocd -s tcl -f tcl/interface/ftdi/esp32_devkitj_v1.cfg -f tcl/board/esp32-wrover.cfg
Open On-Chip Debugger 0.10.0-dev (2018-05-31-23:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
```
**Interesting fact !** After waiting ~30 seconds at this step, the following lines are append to log :
```
Info : Target halted. PRO_CPU: PC=0x400D0A0C (active)    APP_CPU: PC=0x400E16E2 
esp32: target state: halted
```
And the LCD immediately stops blinking. So target seems halted. (Running GDB (step 4.) before or after that event doesnt change anything).

4) Run GDB
```
$ xtensa-esp32-elf-gdb -x gdbinit ./build/blink.elf 
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/blink.elf...done.
timer_process_alarm (dispatch_method=ESP_TIMER_TASK) at /data/user/work/esp/src/esp-idf/components/esp32/esp_timer.c:260
260	{
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
Hardware assisted breakpoint 1 at 0x400d200f: file /data/user/work/esp/src/esp-idf/examples/get-started/blink/main/blink.c, line 43.
0x0:	0x00000000
```
As @Greg-R explained, no GDB prompt here.

Additional OpenOCD log triggered by GDB connection:
```
Info : accepting 'gdb' connection on tcp/3333
Info : Use core0 of target 'esp32'
Info : Target halted. PRO_CPU: PC=0x40091D07 (active)    APP_CPU: PC=0x4000BFDC 
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted

```

igrr, 2018-06-05:
Thanks for the detailed logs @frederic. I'm a bit at loss what might be the reason, since there are no errors reported, but still the breakpoint is not being hit. I have followed the same procedure locally and it seems to work correctly every time.
Here is one more idea to try. Let's see if this is just app_main breakpoint not being hit, or is it the same for any other breakpoint:

1. After doing the steps you did above (and breakpoint on app_main did not hit), hit Ctrl-C. You should get GDB prompt, similar to:
```
^CTarget halted. PRO_CPU: PC=0x400E18BE (active)    APP_CPU: PC=0x400E18BE 
[New Thread 1073438760]
[New Thread 1073442928]
[New Thread 1073441416]
[New Thread 1073429236]
[New Thread 1073412740]
[New Thread 1073413508]

Program received signal SIGINT, Interrupt.
[Switching to Thread 1073436860]
0x400e18be in esp_vApplicationWaitiHook () at /Users/ivan/e/esp-idf/components/esp32/freertos_hooks.c:66
66	    asm("waiti 0");
(gdb) 
```

2. Check the list of tasks using `info threads` command and then switch to "blink_task" task using `thread` command:
```
(gdb) info threads
  Id   Target Id         Frame 
  7    Thread 1073413508 (ipc0) 0x40084719 in xQueueGenericReceive (xQueue=0x3ffafd30, pvBuffer=0x0, 
    xTicksToWait=34402333, xJustPeeking=0) at /Users/ivan/e/esp-idf/components/freertos/queue.c:1591
  6    Thread 1073412740 (esp_timer) 0x40084719 in xQueueGenericReceive (xQueue=0x3ffaea2c, pvBuffer=0x0, 
    xTicksToWait=34402333, xJustPeeking=0) at /Users/ivan/e/esp-idf/components/freertos/queue.c:1591
  5    Thread 1073429236 (ipc1) 0x4008106e in esp_crosscore_int_send_yield (core_id=1)
    at /Users/ivan/e/esp-idf/components/esp32/crosscore_int.c:112
  4    Thread 1073441416 (Tmr Svc) 0x40085edc in prvProcessTimerOrBlockTask (
    xNextExpireTime=<optimized out>, xListWasEmpty=<optimized out>)
    at /Users/ivan/e/esp-idf/components/freertos/timers.c:588
  3    Thread 1073442928 (blink_task) 0x400853be in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/ivan/e/esp-idf/components/freertos/tasks.c:1491
  2    Thread 1073438760 (IDLE : Running @CPU1) 0x400e18be in esp_vApplicationWaitiHook ()
    at /Users/ivan/e/esp-idf/components/esp32/freertos_hooks.c:66
* 1    Thread 1073436860 (IDLE : Running @CPU0) 0x400e18be in esp_vApplicationWaitiHook ()
    at /Users/ivan/e/esp-idf/components/esp32/freertos_hooks.c:66
(gdb) thread 3
[Switching to thread 3 (Thread 1073442928)]
#0  0x400853be in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/ivan/e/esp-idf/components/freertos/tasks.c:1491
1491				portYIELD_WITHIN_API();
(gdb) 
```

3. Get the backtrace:
```
(gdb) bt
#0  0x400853be in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/ivan/e/esp-idf/components/freertos/tasks.c:1491
#1  0x400d2040 in blink_task (pvParameter=0x3ffb0730)
    at /Users/ivan/e/esp-idf/examples/get-started/blink/main/blink.c:34
#2  0x40000001 in ?? ()
(gdb) 
```

4. Now take note of the address in `blink_task` function. In the case above it was `0x400d2040`. Set a breakpoint on *that* address, and resume.

```
(gdb) hb *0x400d2040
Hardware assisted breakpoint 1 at 0x400d2040: file /Users/ivan/e/esp-idf/examples/get-started/blink/main/blink.c, line 36.
(gdb) c
```

5. Check if the breakpoint is hit.
6. Then try using the same address in your gdbinit script, in place of `app_main`. Again, restart GDB to see if this breakpoint is hit.

Greg-R, 2018-06-05:
Here is my gdbinit:
target remote :3333
mon reset halt
thb app_main
x $a1=0
c

Regards,
Greg

Greg-R, 2018-06-05:
I have it on my task list to put a scope on the JTAG.  I have a 4-channel scope with 14MB memory.  This is a new scope, and I have to spend a little time with it understanding how to configure.
I have the working WROOM, so I might be able to do a compare.  Hopefully will get to it within a few days.

I might be able to try the test suggested by Ivan later today.  I will report if possible.

Regards,
Greg

igrr, 2018-06-06:
Okay, that seems to make sense now. PC is 0x40007BA1, which is a location of a while(1) loop in ROM code. ROM code enters this loop if it failed to load application from flash into RAM. If you open UART console before starting OpenOCD, you should see a message like "flash read err" in the console.
Without OpenOCD attached, this infinite loop is usually terminated by RTC watchdog timer, which resets the chip. But when OpenOCD connects to the target, it disables watchdog timers. Therefore once this flash read error happens, the CPU keeps spinning in this infinite loop, and does not reset.
Could you please check the UART output in this situation and confirm whether this is indeed "flash read err". If it is, next step would be to understand why it happens.

By the way, you mentioned that you are using ESP-WROVER-KIT. Is this [V2 board](http://esp-idf.readthedocs.io/en/latest/hw-reference/modules-and-boards-previous.html#esp-wrover-kit-v2) (glossy black solder mask, male connector for camera) or [V3 board](http://esp-idf.readthedocs.io/en/latest/hw-reference/modules-and-boards.html#esp-wrover-kit-v3) (matte black solder mask, female connector for camera)? Also, what module is installed on the board, ESP-WROOM32 or ESP32-WROVER? Usually V2 will have WROOM and V3 will have WROVER, but i want to confirm this.

Greg-R, 2018-06-06:
Copied below is what I see with "make monitor".  Looks good to me.  This is the matte black board with ESP32-WROVER module.  The only female connector is labelled JP4.  The "blink" example toggles GPIO5, which in the case of the WROVER kit, causes the entire display to blink.  So that makes me believe the program is loaded up correctly and is running.  The power is via USB.  Could there be excessive voltage drop due to the current drain from the display?  I'll try moving the GPIO to a different one (not one of the JTAGs) later today and see if it makes a difference.
Here is the monitor output:
blink$ make monitor
MONITOR
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5780
load:0x40078000,len:0
load:0x40078000,len:14960
entry 0x4007862c
I (28) boot: ESP-IDF v3.1-dev-1193-g64b56bef 2nd stage bootloader
I (28) boot: compile time 16:59:11
I (30) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05a94 ( 23188) map
I (101) esp_image: segment 1: paddr=0x00015abc vaddr=0x3ffb0000 size=0x022cc (  8908) load
I (105) esp_image: segment 2: paddr=0x00017d90 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /mnt/ds216/embedded_design/espressif/esp-idf/components/freertos/xtensa_vectors.S:1685

I (111) esp_image: segment 3: paddr=0x00018198 vaddr=0x40080400 size=0x07e78 ( 32376) load
I (133) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x11c74 ( 72820) map
0x400d0018: _flash_cache_start at ??:?

I (159) esp_image: segment 5: paddr=0x00031c94 vaddr=0x40088278 size=0x008b8 (  2232) load
0x40088278: esp_rom_spiflash_read_status at /mnt/ds216/embedded_design/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

I (160) esp_image: segment 6: paddr=0x00032554 vaddr=0x400c0000 size=0x00000 (     0) load
I (171) boot: Loaded app from partition at offset 0x10000
I (172) boot: Disabling RNG early entropy source...
I (177) cpu_start: Pro cpu up.
I (181) cpu_start: Starting app cpu, entry point is 0x40080e48
0x40080e48: call_start_cpu1 at /mnt/ds216/embedded_design/espressif/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (191) heap_init: Initializing. RAM available for dynamic allocation:
I (198) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (204) heap_init: At 3FFB3300 len 0002CD00 (179 KiB): DRAM
I (210) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (217) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (223) heap_init: At 40088B30 len 000174D0 (93 KiB): IRAM
I (229) cpu_start: Pro cpu start user code
I (247) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

igrr, 2018-06-06:
Is this this output you get right after GDB starts up and executes the startup script? I understand that example runs correctly when you reset the board using 'make monitor'. What seems to happen is, when OpenOCD resets the chip (`monitor reset`), then the ROM code can not load the application.

Please try opening `make monitor` first, don't close it, then in another console start GDB. Observe what is printed to UART.

Greg-R, 2018-06-06:
OK, very interesting, your suggestion worked!  So I did this:
1.  make monitor in blink project.  See terminal output above.
2.  Start Openocd in 2nd terminal.
3.  In 3rd terminal, start GDB.

Now there is additional output from monitor, append this to the make monitor output from above:
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

Now, where do we go from here?

frederic, 2018-06-08:
Just to confirm I have the exact same results than @Greg-R : when I interrupt GDB, and when I use monitor + openocd + GDB.

frederic, 2018-06-08:
So without using the gdbinit file, it seems to works better:
0) flash blink example & boot
1) Attach OpenOCD
2) Run GDB with blink.elf & and execute `target remote :3333`

```
(gdb) target remote :3333
Remote debugging using :3333
0x400e16e2 in esp_vApplicationWaitiHook () at /data/user/work/esp/src/esp-idf/components/esp32/freertos_hooks.c:66
66	    asm("waiti 0");
(gdb) info thread
[New Thread 1073436868]
[New Thread 1073442952]
[New Thread 1073441424]
[New Thread 1073429244]
[New Thread 1073412740]
[New Thread 1073413508]
  Id   Target Id         Frame 
  7    Thread 1073413508 (ipc0) 0x40084585 in xQueueGenericReceive (xQueue=0x3ffafd30, pvBuffer=0x0, xTicksToWait=1644638200, xJustPeeking=0) at /data/user/work/esp/src/esp-idf/components/freertos/queue.c:1591
  6    Thread 1073412740 (esp_timer) 0x40081812 in esp_crosscore_int_send_yield (core_id=0) at /data/user/work/esp/src/esp-idf/components/esp32/crosscore_int.c:112
  5    Thread 1073429244 (ipc1) 0x40081812 in esp_crosscore_int_send_yield (core_id=1) at /data/user/work/esp/src/esp-idf/components/esp32/crosscore_int.c:112
  4    Thread 1073441424 (Tmr Svc) 0x40085f34 in prvProcessTimerOrBlockTask (xNextExpireTime=<optimized out>, xListWasEmpty=<optimized out>) at /data/user/work/esp/src/esp-idf/components/freertos/timers.c:588
  3    Thread 1073442952 (blink_task) 0x40085416 in vTaskDelay (xTicksToDelay=<optimized out>) at /data/user/work/esp/src/esp-idf/components/freertos/tasks.c:1491
  2    Thread 1073436868 (IDLE : Running @CPU0) 0x400e16e2 in esp_vApplicationWaitiHook () at /data/user/work/esp/src/esp-idf/components/esp32/freertos_hooks.c:66

The current thread <Thread ID 1> has terminated.  See `help thread'.
(gdb) thread 3
[Switching to thread 3 (Thread 1073442952)]
#0  0x40085416 in vTaskDelay (xTicksToDelay=<optimized out>) at /data/user/work/esp/src/esp-idf/components/freertos/tasks.c:1491
1491				portYIELD_WITHIN_API();
(gdb) bt
#0  0x40085416 in vTaskDelay (xTicksToDelay=<optimized out>) at /data/user/work/esp/src/esp-idf/components/freertos/tasks.c:1491
#1  0x400d1ffc in blink_task (pvParameter=0x3ffb1570) at /data/user/work/esp/src/esp-idf/examples/get-started/blink/main/blink.c:34
#2  0x40000001 in ?? ()
(gdb) hb *0x400d1ffc
Hardware assisted breakpoint 1 at 0x400d1ffc: file /data/user/work/esp/src/esp-idf/examples/get-started/blink/main/blink.c, line 36.
(gdb) c
Continuing.
[New Thread 1073438768]

Program received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073436868]
0x400e16e8 in heap_caps_match (heap=0x8, caps=0) at /data/user/work/esp/src/esp-idf/components/heap/heap_caps.c:60
60	    return heap->heap != NULL && ((get_all_caps(heap) & caps) == caps);
(gdb) c
Continuing.
Target halted. PRO_CPU: PC=0x400D0A0C (active)    APP_CPU: PC=0x400E16E2 

Program received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073412740]
timer_process_alarm (dispatch_method=ESP_TIMER_TASK) at /data/user/work/esp/src/esp-idf/components/esp32/esp_timer.c:260
260	{
(gdb) c
```
However the breakpoint on blink_task (0x400d1ffc) is never hit, not sure if that's expected.
Instead, I have the SIGTRAP exception that keeps interrupting.

frederic, 2018-06-08:
FYI, when I execute gdbinit commands manually :
```
$ xtensa-esp32-elf-gdb  ./build/blink.elf 
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/blink.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
0x400e16e2 in esp_vApplicationWaitiHook () at /data/user/work/esp/src/esp-idf/components/esp32/freertos_hooks.c:66
66	    asm("waiti 0");
(gdb) mon reset halt
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
(gdb) info threads
  Id   Target Id         Frame 
* 1    Remote target     0x400e16e2 in esp_vApplicationWaitiHook () at /data/user/work/esp/src/esp-idf/components/esp32/freertos_hooks.c:66
(gdb) thb app_main
Hardware assisted breakpoint 1 at 0x400d200f: file /data/user/work/esp/src/esp-idf/examples/get-started/blink/main/blink.c, line 43.
(gdb) x $a1=0
0x0:	0x00000000
(gdb) c
Continuing.
^CTarget halted. PRO_CPU: PC=0x40007BA1 (active)    APP_CPU: PC=0x400076E2 

Program received signal SIGINT, Interrupt.
0x40007ba1 in ?? ()
(gdb) info threads
  Id   Target Id         Frame 
* 1    Remote target     0x40007ba1 in ?? ()
(gdb) bt
#0  0x40007ba1 in ?? ()
#1  0x40000740 in ?? ()
```
Interesting to note that when I enter the continue (c) command in GDB, the LCD doesnt resume blinking (stays off).

igrr, 2018-06-08:
@frederic could you try running the latest test, where you executed gdbinit commands manually, but use `flushregs` instead of `x $a1=0`?

frederic, 2018-06-08:
Here it is :
```
$ xtensa-esp32-elf-gdb  ./build/blink.elf 
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/blink.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
0x400e16e2 in esp_vApplicationWaitiHook () at /data/user/work/esp/src/esp-idf/components/esp32/freertos_hooks.c:66
66	    asm("waiti 0");
(gdb) mon reset halt
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
(gdb) thb app_main
Hardware assisted breakpoint 1 at 0x400d200f: file /data/user/work/esp/src/esp-idf/examples/get-started/blink/main/blink.c, line 43.
(gdb) flushregs
Register cache flushed.
(gdb) c
Continuing.
^CTarget halted. PRO_CPU: PC=0x40007BA1 (active)    APP_CPU: PC=0x400076DD 

Program received signal SIGINT, Interrupt.
0x40007ba1 in ?? ()
(gdb) info threads
  Id   Target Id         Frame 
* 1    Remote target     0x40007ba1 in ?? ()
(gdb) bt
#0  0x40007ba1 in ?? ()
#1  0x40000740 in ?? ()
```
Same issue with LCD, it doesnt resume blinking.

igrr, 2018-06-08:
At the prompt you get after pressing Ctrl-C, could you please run the following command and paste the output here?

```
x/1xw 0x3ff48074
```

frederic, 2018-06-08:
(gdb) x/1xw 0x3ff48074
0x3ff48074:	0xff600000

frederic, 2018-06-08:
I noticed a pretty concerning issue too : when I disassemble a function in GDB, output keeps changing each time I reboot :

```
(gdb) disassemble gpio_set_level 
Dump of assembler code for function gpio_set_level:
   0x400e1158 <+0>:	entry	a1, 48
   0x400e115b <+3>:	movi.n	a8, 39
[...]
   0x400e11fe <+166>:	l32i.n	a9, a8, 24
   0x400e1200 <+168>:	clrb_expstate	0
   0x400e1203 <+171>:	f64cmph	a0, a0, a4, 15
   0x400e1206 <+174>:	f64cmph	a0, a0, a0, 7
   0x400e1209 <+177>:	ill
   0x400e120c <+180>:	f64cmph	a0, a7, a0, 15
   0x400e120f <+183>:	bnez.n	a0, 0x400e1243 <gpio_set_direction+43>
   0x400e1211 <+185>:	rsr.lbeg	a0
   0x400e1214 <+188>:	callx12	a0
End of assembler dump.
```
*reset*
```
(gdb) disassemble gpio_set_level 
Dump of assembler code for function gpio_set_level:
   0x400e1158 <+0>:	entry	a1, 48
   0x400e115b <+3>:	movi.n	a8, 39
[...]
 0x400e11fe <+166>:	l32i.n	a9, a8, 24
   0x400e1200 <+168>:	clrb_expstate	0
   0x400e1203 <+171>:	f64iter	a0, a0, a4, 3, 1
   0x400e1206 <+174>:	f64cmph	a0, a0, a0, 7
   0x400e1209 <+177>:	lsxp	f1, a8, a6
   0x400e120c <+180>:	bnei	a7, -1, 0x400e120f <gpio_set_level+183>
   0x400e120f <+183>:	.byte 0xfe
   0x400e1210 <+184>:	ill
   0x400e1213 <+187>:	isync
End of assembler dump.
```
*reset*
```
(gdb) disassemble gpio_set_level 
Dump of assembler code for function gpio_set_level:
   0x400e1158 <+0>:	entry	a1, 48
   0x400e115b <+3>:	movi.n	a8, 39
[...]
   0x400e11fe <+166>:	l32i.n	a9, a8, 24
   0x400e1200 <+168>:	clrb_expstate	0
   0x400e1203 <+171>:	f64cmph	a0, a15, a4, 15
   0x400e1206 <+174>:	f64cmph	a0, a0, a0, 15
   0x400e1209 <+177>:	rfe
   0x400e120c <+180>:	f64cmph	a0, a7, a0, 15
   0x400e120f <+183>:	bnez.n	a0, 0x400e1243 <gpio_set_direction+43>
   0x400e1211 <+185>:	rsr.lbeg	a0
   0x400e1214 <+188>:	callx12	a0
End of assembler dump.
```

Only the end of function seems to be affected.

igrr, 2018-06-08:
> when I disassemble a function in GDB, output keeps changing each time I reboot

This is probably because after reset, flash cache is disabled, but OpenOCD is trying to read the code from the cached region (it doesn't know that app has not enabled flash cache yet). So it reads garbage instead.

Regarding the actual issue (ROM code can't read flash following restart) we're still pondering what may be the cause. I'll try to give some update soon.

frederic, 2018-06-20:
Hello,

I am still stuck with this issue.
I was wondering if the builtin JTAG of ESP-WROVER-KIT V3 works for you guys at Expressif, and in such case what setup (host, idf version, openocd version) do you use ?

Thank you
Regards,
Frederic

yaoyogi, 2019-01-28:
I had a similar problem with (gdb) info threads - showing no threads.
What fixed the situation for me was to make sure that the menuconfig "Run FreeRTOS" only on first core and your openocd board config agree on whether you are running single core or multi-core.  Once I got those settings in agreement (both single core OR both dual core)  I didn't run into the "no threads" problem in (gdb)

KampfCaspar, 2019-04-11:
@yaoyogi Unfortunately, neither dual core nor single core (corresponding freertos and openocd board config) work with me... Issue persists. :(

cooperbaker, 2019-04-29:
Similar issue here.  GDB successfully connects to OpenOCD about 1/3 of the time.  I have to repeatedly try to connect gdb and eventually it will 'work'.

Unsuccessful Attempts:
GDB : ERROR: Unable to start debugging. Failed to find thread 3 for break event
OPENOCD: Info : dropped 'gdb' connection

I'm trying to set up a workflow for end users to develop on a product using esp32... Would really like to solve this and am happy to provide logs/info if anyone who knows has time to help.

gerekon, 2021-01-11:
Guys, If it is still actual. Could you check if problem still exists with the latest OpenOCD release (https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20201202)? We've made huge refactoring and applied many fixes since that time.