espressif/openocd-esp32 #40: GDB access to CPU1

simon-jouet, 2018-06-09:
Hi,

After a bit of struggle I have managed to get openocd to connect properly to the ESP32 and I can connect gdb to port 3333 without any issue, listing the threads setting breakpoints ...

The issue is with connect the the APP CPU, the esp-idf documentation details that port 3333 is for CPU0 and port 3334 for CPU1, however when I run openocd it doesn't bind on port 3334. The issue is I think fairly similar to the last post on this thread: https://www.esp32.com/viewtopic.php?f=13&t=2497&p=13035&hilit=3334#p13035

I'm running the latest release of openocd-esp32 with a FT2232H as a jtag debugger and using the default configuration as shown below. The only modification I've made is to slow down the adapter speed (at least for now) because i'm using dupont cables to connect the ESP32 to the ft2232.

```bash
$ bin/openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg

Open On-Chip Debugger 0.10.0-dev (2018-06-04-09:51)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
esp32 interrupt mask on
force hard breakpoints
Info : clock speed 5000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
```

in gdb
```bash
(gdb) target remote :3333
Remote debugging using :3333
0x4014d522 in esp_vApplicationWaitiHook ()
    at C:/msys32/home/simon/esp/esp-idf/components/esp32/freertos_hooks.c:66
66          asm("waiti 0");
(gdb) mon reset
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).

(gdb) target remote localhost:3334
localhost:3334: The system tried to join a drive to a directory on a joined drive.

```

I've spent quite a bit of time going through the forum, the doc and the different config files to see if I was missing something obvious but I can't seem to figure it out.

Best regards,
Simon

igrr, 2018-06-09:
If you run OpenOCD as you have shown, there will be no access to individual
CPUs. Instead, FreeRTOS support is enabled, and tasks running on both CPUs
are listed in GDB as "threads". Switching between tasks allows you to
change the active CPU used for single stepping.

If you wish to disable RTOS support and access CPUs using two GDB instances
(one instance can not connect to two targets), you need to modify esp32.cfg
file.

Also, could you please point which documentation page describes connecting
to port 3334? We'll make sure it is updated and points to latest
instructions in the IDF programming guide. Thanks.

simon-jouet, 2018-06-09:
Hi @igrr 

Thanks for the very fast reply and the details, I didn't get that 3334 was disabled when FreeRTOS support is enabled. I'm trying to debug a WDT interrupt timeout and getting the debugger working should help.

Regarding the doc I mostly followed https://esp-idf.readthedocs.io/en/v2.0/openocd.html which is for the previous version of the esp-idf, but I didn't realise that JTAG debugging also changed. Anyway thanks very much for the help -- sorry this was more support than an issue with openocd-esp32 and should probably have been on the forum instead of here

simon-jouet, 2018-06-09:
Thanks for the details, I've managed to debug the issue! The watchdog timeout being disabled didn't end up being too much an issue as I could just halt the CPU when it hanged and figured from the backtrace where the code was.

I tracked the issue to my code calling `timer_enable_intr()` and the code hanging while calling that - I haven't figure the reason why it hanged just yet - but by commenting this out for the time being it looks like it's working.

Anyway thanks again, having the jtag for debugging will definitely help a lot.

Cheers,
Simon

closed 2018-06-09 by simon-jouet