espressif/openocd-esp32 #42: "nexti" does not step over "call0" instruction

michprev, 2018-06-14:
When debugging in GDB "nexti" (next instruction) command does not step over "call0" instruction but steps inside. Tried to delete all breakpoints before with no help.

Using OpenOCD v0.10.0-esp32-20180604.

michprev, 2018-08-26:
Problem still persists in v0.10.0-esp32-20180724.

A simple example:
```
    .org    0x1C0
    .section .vectors
    .global _Level3InterruptVector
    .type   _Level3InterruptVector,@function
    .align 4
_Level3InterruptVector:

    // make a backup of a0
    wsr.excsave3    a0
    // call0 used as jump
    call0           _Level3Interrupt

/*
--------------------------------------------------------------------------------
*/

    .section .iram
    .global _Level3Interrupt
    .type   _Level3Interrupt,@function
    .align 4
_Level3Interrupt:

    mov             a0, sp                      // backup stack pointer before interrupt
    addi            sp, sp, -XT_STK_FRMSZ       // move stack pointer to create a new stack frame
    s32i            a0, sp, XT_STK_A1           // save pre-interrupt stack pointer on the stack

    call0           interrupt_save_context
```

GDB commands to reproduce:
```
b _Level3InterruptVector
ni
si # step into _Level3Interrupt
ni
ni
ni
ni # should step over interrupt_save_context but steps inside
```

erhankur, 2023-12-13:
@michprev We are sorry to leave this issue without a response.  Now, closing this one and if you still have an issue with the latest OpenOCD, esp-idf, Let us know.

closed 2023-12-13 by erhankur