espressif/openocd-esp32 #49: Could not upload program in GDB

tuangu, 2018-07-05:
OpenOCD version: v0.10.0-esp32-20180604
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 7.10

I am facing a problem with uploading code to target (ESP-WROVER-KIT v3) under gdb using program_esp32 tool. It keeps saying "couldn't open [filename]". I have tried using OpenOCD command and it also did not work. Both produce the same error message.

gdb log:
```
(gdb) mon reset halt
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
esp32: target state: halted
(gdb) pwd
Working directory C:\msys32\home\Tuan\esp\blink\build.
(gdb) mon program_esp32 blink.bin 0x10000
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
esp32: target state: halted
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
esp32: target state: halted
** Programming Started **
auto erase enabled
couldn't open blink.bin
embedded:startup.tcl:454: Error: ** Programming Failed **
in procedure 'program_esp32'
in procedure 'program_error' called at file "share/openocd/scripts/target/esp32.cfg", line 155
at file "embedded:startup.tcl", line 454
```

OpenOCD command:
```
bin/openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 /home/Tuan/esp/blink/build/blink.bin 0x10000 reset"
```

worker2345234, 2018-07-05:
try `"program_esp32 C:/msys32/home/Tuan/esp/blink/build/blink.bin 0x10000 reset"`

tuangu, 2018-07-05:
@worker2345234  Thanks for your reply. Try your solution and the problem solved.

closed 2018-07-05 by tuangu