espressif/openocd-esp32 #51: Can not set software breakpoints(more than two breakpoints)

dzRBRglobal, 2018-07-27:
JTAG adapter type: FT2232H on ESP-WROVER-KIT
    ESP-IDF : v3.2-dev-209-gbe81d2c1
    OS : Ubuntu Ubuntu 18.04.1 LTS
    Is OS running natively on a PC or on a virtual machine? On Virtual box

Can not set software breakpoints(more than two breakpoints), please see attached logs for details.
[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/2236886/openocd_log.txt)
[gdb_log.txt](https://github.com/espressif/openocd-esp32/files/2236887/gdb_log.txt)

gerekon, 2018-07-30:
Hi @dzRBRglobal 

What version of OpenOCD do you use? I checked your case with the latest release from https://github.com/espressif/openocd-esp32/releases and it works fine for me.

ivankravets, 2018-07-30:
The same for me, `thb app_main` has stopped working with the last release where were introduced soft breakpoints. Mac.

igrr, 2018-07-30:
@ivankravets Are you seeing the same error as reported by @dzRBRglobal above? Please share your logs as well: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-reporting-issues.

ivankravets, 2018-07-30:
@igrr see a full log https://community.platformio.org/t/esp32-pio-unified-debugger/4541/21

gerekon, 2018-07-30:
@ivankravets This seems to be different issue. I will have a look at it.

gerekon, 2018-07-30:
@ivankravets 
Hmm, I checked `thb app_main` with blink example and the latest release of openocd. It works. Could you post full log of GDB and OpenOCD using [this instructions](https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-reporting-issues)?
Also could you post symbol map of your application?

gerekon, 2018-07-30:
@ivankravets Do you have partition table at the standard offeset (0x8000) in flash? It seems that OOCD can not read flash mappings for your application.

ivankravets, 2018-07-30:
I'll provide more details soon. We are going to release the next update to esp32 dev/platform today. We will stay with the previous release. It works without any issues on Mac/Win/Linux. Many developers confirmed that with 1-click debugging using Arduino or ESP-IDF frameworks.

You can try PlatformIO in pair with VSCode and replace the default `openocd-esp32` package by yours in `~/.platformio/packages/tool-openocd-esp32`.

dzRBRglobal, 2018-07-30:
Hi @gerekon 
Thank you very much for you immediate reply. Just looked my OpenOCD version, it is 0.10.0-dev (2018-06-04-09:50)

gerekon, 2018-07-30:
@dzRBRglobal 

> Just looked my OpenOCD version, it is 0.10.0-dev (2018-06-04-09:50)

It is old version. Should be from 2018-07-24.

dzRBRglobal, 2018-07-30:
@gerekon 
I just tried 2018-07-24. The problem is fixed. Thank you.

closed 2018-08-16 by igrr