espressif/openocd-esp32 #52: Error programming ESP32 with 16MB flash

sinope-db, 2018-07-31:
I've just switch to ESP-WROOM-32 modules with 16MB of flash and I'm getting an error when trying to program them using JTAG. Using the same setup, I can program modules with 4MB flash just fine.

I'm using the win32 binaries of the release v0.10.0-esp32-20180724.
I also tried the previous release (20180604) without success.

> OpenOCD_esp20180724>bin\openocd -s share\openocd\scripts -f interface/jlink.cfg -f target/esp32.cfg -c "adapter_khz 3000" -c "program_esp32 bootloader.bin 0x1000 verify exit"

> Open On-Chip Debugger 0.10.0-dev (2018-07-24-09:39)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
esp32 interrupt mask on
adapter speed: 3000 kHz
Info : No device selected, using first device.
Info : J-Link V10 compiled Oct  9 2015 20:36:39
Info : Hardware version: 10.10
Info : VTarget = 3.322 V
Info : clock speed 3000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x40000400
Error: Failed to get flash maps (-1)!
Error: Failed to get flash mappings (-4)!
Error: auto_probe failed
** Programming Failed **
shutdown command invoked

[openocd_16MB.log](https://github.com/espressif/openocd-esp32/files/2246681/openocd_16MB.log)

gerekon, 2018-08-09:
@sinope-db Could you also post your partition table?

sinope-db, 2018-08-10:
I tried programming bootloader and main app addresses and both failed 
It's also failing whether module is brand new or has been programmed using the serial bootloader.

My partitions (ESP-IDF 3.0.2):
```
# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,        data, nvs,     0x9000,   0x4000,
otadata,    data, ota,     0xd000,   0x2000,
phy_init,   data, phy,     0xf000,   0x1000,
AtpData,    0x40, 0,       0x10000,  0x1000,
dynamic_fs, data, spiffs,  0x20000,  0x20000,
static_fs,  data, spiffs,  0x40000,  0x20000,
factory,    app,  factory, 0x400000, 4M,
ota_0,      app,  ota_0,   ,         4M,
ota_1,      app,  ota_1,   ,         4M,
```

[partitions.zip](https://github.com/espressif/openocd-esp32/files/2277900/partitions.zip)

gerekon, 2018-08-10:
@sinope-db Unfortunately I have no 16 MB module at hands right now. I am trying to get it ASAP.
Just to check. Could you try to move factory partition to standard location at 0x10000?
Also could you attach your sdkconfig and full OpenOCD+GDB log (use [this instructions)](https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-reporting-issues)?

sinope-db, 2018-08-14:
@gerekon Moving the factory partition does not solve the issue.

OpenOcd log is attached to OP, but I'm attaching one with the partition set at 0x10000. [openocd_log.txt](https://github.com/espressif/openocd-esp32/files/2286939/openocd_log.txt)

As for GDB log, I'm not using GDB at this point, only flashing according to these [instructions]
(https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/index.html#upload-application-for-debugging).

sorenhl, 2018-09-12:
Any updates on this one? I face the same issue

sorenhl, 2018-09-12:
Just an update, when I try to use my ESP-WROOM-32 developer board, where i run the firmware on a 16 MB external flash (using eFuses) it works. But when I use my manufactured PCB, where the ESP-WROOM-32D comes with 16MB, it fails:

Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
esp32 interrupt mask on
adapter speed: 3000 kHz
VisualGDB_OpenOCD_Ready
Info : clock speed 3000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x4000C02B (active)    APP_CPU: PC=0x4016810E 
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : accepting 'gdb' connection on tcp/22402
Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x4016810E 
Error: Failed to get flash maps (-1)!
Error: Failed to get flash mappings (-4)!
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

gerekon, 2018-09-13:
@sinope-db @sorenhl I have got the board. I will try to solve the problem next week.

sorenhl, 2018-09-13:
Thank you @gerekon, that would be really appreciated!

closed 2018-09-19 by igrr

SunilDullolli, 2018-09-19:
@igrr can you release new binaries for updated OpenOCD? I am having hard time to compile OpenOCD because of architecture mismatches. Thanks

igrr, 2018-09-20:
@SunilDullolli done, please check https://github.com/espressif/openocd-esp32/releases

sorenhl, 2018-09-20:
Thank you so much @igrr and @gerekon, I have tested on my latest PCB with a 16MB ESP-WROOM-32D and I can confirm that it works as intended :)