espressif/openocd-esp32 #56: After flashing once over JTAG, successive attempts fail

LaurentLouf, 2018-08-03:
Hello,

I'm using an Olimex ARM-USB-OCD-H to flash an ESP32 (ESP32 WROOM 32U) with OpenOCD ESP32 (Open On-Chip Debugger 0.10.0-dev (2018-07-27-18:59)) with the following scripts (in no particular order) : 
`program_esp32 build/app.bin 0x10000 verify exit`
```
source [find interface/ftdi/olimex-arm-usb-ocd-h.cfg]
transport select jtag
source [find target/esp32.cfg]
adapter_khz 200
```
And the command : 
`openocd-esp32 -s /usr/local/openocd-esp32/share/openocd/scripts/ -f $cfgpath -f openocd_esp32_flash.cfg`

The first time I tried, I had the wiring wrong (TDI and TDO), so I fixed that and retried. To add some info, IO0 is connected to ground. And all went well (the verify part also) : 

> Open On-Chip Debugger 0.10.0-dev (2018-07-27-18:59)
> Licensed under GNU GPL v2
> For bug reports, read
>     http://openocd.org/doc/doxygen/bugs.html
> esp32 interrupt mask on
> adapter speed: 200 kHz
> Info : clock speed 200 kHz
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> cpu0: Current bits set: BreakIn BreakOut RunStallIn
> cpu1: Current bits set: none
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> cpu0: Current bits set: BreakIn BreakOut RunStallIn
> cpu1: Current bits set: none
> Info : Target halted. PRO_CPU: PC=0x40009051 (active)    APP_CPU: PC=0x00000000 
> Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
> Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
> Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
> Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
> Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
> ** Programming Started **
> auto erase enabled
> Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x40000400 
> Info : Flash mapping 0: 0x110010 -> 0x3f400010, 113 KB
> Info : Flash mapping 1: 0x150018 -> 0x400d0018, 489 KB
> Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x40000400 
> Info : Auto-detected flash size 4096 KB
> Info : Using flash size 4096 KB
> Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x40000400 
> Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x40000400 
> wrote 1433600 bytes from file build/app.bin in 161.379593s (8.675 KiB/s)
> ** Programming Finished **

Since flashing and verifying was a bit slow, I decided to retry with higher values for `adapter_khz` , without much success. So I went back to a speed of 200kHz, since it worked previously, but now, all I get is : 

> Open On-Chip Debugger 0.10.0-dev (2018-07-27-18:59)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> esp32 interrupt mask on
> adapter speed: 200 kHz
> Info : clock speed 200 kHz
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
> cpu0: Current bits set: BreakIn BreakOut RunStallIn
> cpu1: Current bits set: BreakIn BreakOut RunStallIn
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> cpu0: Current bits set: BreakIn BreakOut RunStallIn
> cpu1: Current bits set: BreakIn BreakOut RunStallIn
> Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
> Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
> Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
> Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
> Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
> ** Programming Started **
> auto erase enabled
> Info : Target halted. PRO_CPU: PC=0x4009170E (active)    APP_CPU: PC=0x40000400 
> Info : Flash mapping 0: 0x751b9fce -> 0x6568dbee, 3050264 KB
> Info : Flash mapping 1: 0xf3bddf15 -> 0x76894cb5, 2658704 KB
> Info : Flash mapping 2: 0x0 -> 0x78c9eb90, 21 KB
> Info : Flash mapping 3: 0xc55d4a00 -> 0x7c76f945, 1978865 KB
> Info : Flash mapping 4: 0x5652 -> 0x10000, 0 KB
> Info : Flash mapping 5: 0x78c6c410 -> 0x5652, 2040059 KB
> [...]
> Info : Flash mapping 1416: 0x6c61636f -> 0x6e69622f, 1661979 KB
> Info : Flash mapping 1417: 0x64636f6e -> 0x7073652d, 770060 KB
> Info : Flash mapping 1418: 0x2f727375 -> 0x61636f6c, 1726603 KB
> Info : Flash mapping 1419: 0x706f2f6e -> 0x636f6e65, 1890635 KB
> Info : Flash mapping 1420: 0x323370 -> 0x0, 0 KB
> ./build.sh: line 246: 12783 Segmentation fault      (core dumped) openocd-esp32 -s /usr/local/openocd-esp32/share/openocd/scripts/ -f $cfgpath -f openocd_esp32_flash.cfg

The number of lines containing "Flash mapping" varies from time to time. So I guess it's not really the seg fault the problem here, but that it happens because of the wrong readings of the flash mapping. But then, I have no idea why these are wrong, while they were good a few minutes before. Any ideas ?

gerekon, 2018-08-10:
@LaurentLouf Yes, the problem is not in seg fault (BTW, it should be fixed in the next release).
Could you try to read 4096 KB at addresses 0x8000 and 0x10000 and post the contents? If the [latest version](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20180724) of OpenOCD does not work for this could you try it with the [previous one](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20180604)?
Could you also provide your partition table?

igrr, 2018-11-22:
Closing due to lack of feedback, please reopen if the issue is still valid.

closed 2018-11-22 by igrr