espressif/openocd-esp32 #58: [TW#25931] Issue with running gdb on ESP32 Wrover Dev Kit

igrr, 2018-09-05:
_From @RA5040 on August 25, 2018 17:34_

## Environment

- Development Kit:   ESP32-Wrover-Kit

- esp-idf-v3.0.3

 - openocd-esp32-win32-0.10.0-esp32-20180724

- Development Env:		Command line

- Operating System:	Windows

## Problem Description

It appears that to run gdb on the ESP32 DevKit C, it's necessary to do things in this order:

1. Make and flash the program
2. Run OpenOCD
3. Run xtensa-esp32-elf-gdb

If OpenOCD is running before flashing the program nothing works.

The Espressif documentation says that OpenOCD should be running, then the program should be flashed, then gdb should be run: https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-command-line

So there's either a bug in the code or a bug in the documentation

### Expected Behavior

OpenOCD should be running before flashing and starting gdb

### Actual Behavior

If OpenOCD is running the gdb fails

### Steps to reproduce

Using a ESP32WroverKit with built-in JTAG, follow the Espressif instructions to debug with JTAG in Windows to the letter, build and flash one of the IDF examples (for example blink), and you should see that dbg does not work.

Flash the program, then run OpenOCD, then run the xtensa gdb and hardware debugging should work.

Here is the gdb log:
===============
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-host_pc-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...done.
0x400d1ffa in app_main () at C:/msys32/home/Robert/esp/blink/main/blink.c:45
45 vTaskDelay(1000);
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active) APP_CPU: PC=0x00000000
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active) APP_CPU: PC=0x40000400
[... 133 lines cut ...]
34	vTaskDelay(1000 / portTICK_PERIOD_MS);
35	/* Blink on (output high) */
36	gpio_set_level(BLINK_GPIO, 1);
37	vTaskDelay(1000 / portTICK_PERIOD_MS);
38	}
39	}
40	
41	void app_main()
42	{
43	xTaskCreate(&blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
44	while(true){
45	vTaskDelay(1000);
46	}
47	}

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/2320892/sdkconfig.zip)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[blink.zip](https://github.com/espressif/esp-idf/files/2320891/blink.zip)

_Copied from original issue: espressif/esp-idf#2332_

igrr, 2018-09-05:
_From @RA5040 on August 26, 2018 11:12_

I was asked to run this again with -d3 on openocd. I attach the log.
[OpenOCDLog.zip](https://github.com/espressif/esp-idf/files/2321462/OpenOCDLog.zip)

gerekon, 2018-09-14:
@RA5040 I reproduced you this problem with the latest release. But it seems to be solved in the OpenOCD master branch. I built Windows binary for you using that branch. Could you try it?
[openocd.zip](https://github.com/espressif/openocd-esp32/files/2382829/openocd.zip)

RA5040, 2018-09-14:
I gave it a go, but it is missing libusb-1.0.dll
I tried it again on ubuntu, openocd 0.10.0-dev (2018-07-24-09:36), but it gave the same issue.  If you let me have the executable I'll test it (I don't know how to build it).

gerekon, 2018-09-18:
@RA5040 Sorry for inconvenience. Could you try this distro. It is build by our release system and should have no problems.
[openocd-esp32-win32-0.10.0-esp32-20180724-2-g59d39ac2.zip](https://github.com/espressif/openocd-esp32/files/2391132/openocd-esp32-win32-0.10.0-esp32-20180724-2-g59d39ac2.zip)

kangks, 2018-11-04:
Hi,

Not sure if my issue is related to this, I was following the steps in https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/jtag-debugging/using-debugger.rst

After running `xtensa-esp32-elf-gdb -x gdbinit build/blink.elf`, the LCD stops blinking for a while, but resume blinking after the last line:

```
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
Hardware assisted breakpoint 1 at 0x400d1fdb: file /Users/kangrich/Documents/tools/esp-idf-v3.0.5/examples/get-started/blink/main/./blink.c, line 43.
```

gdb and openocd as below, on MAC High Sierrra:

```
$ xtensa-esp32-elf-gdb --version
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

$ bin/openocd -v
Open On-Chip Debugger 0.10.0-dev (2018-09-20-12:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

kangks, 2018-11-04:
Built from the openocd works:

```
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_apple-darwin16.3.0 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...done.
0x400d1afe in esp_vApplicationIdleHook () at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/esp32/./freertos_hooks.c:62
62	    asm("waiti 0");
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
Hardware assisted breakpoint 1 at 0x400d1fdb: file /Users/kangrich/Documents/tools/esp-idf-v3.0.5/examples/get-started/blink/main/./blink.c, line 43.
Target halted. PRO_CPU: PC=0x400D1FDB (active)    APP_CPU: PC=0x400D1AFE 
[New Thread 1073435424]
[New Thread 1073434036]
[New Thread 1073438080]
[New Thread 1073426924]
[New Thread 1073412716]
[New Thread 1073413484]
[New Thread 1073428056]
[Switching to Thread 1073432516]

Temporary breakpoint 1, app_main () at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/examples/get-started/blink/main/./blink.c:43
43	    xTaskCreate(&blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
(gdb) i threads
  Id   Target Id         Frame 
  8    Thread 1073428056 (dport) xTaskCreate (pvCreatedTask=0x0, uxPriority=5, pvParameters=0x0, usStackDepth=768, pcName=0x3f402ba8 "blink_task", 
    pvTaskCode=0x400d1fa8 <blink_task>) at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/freertos/include/freertos/task.h:427
  7    Thread 1073413484 (ipc0) xTaskCreate (pvCreatedTask=0x0, uxPriority=5, pvParameters=0x0, usStackDepth=768, pcName=0x3f402ba8 "blink_task", 
    pvTaskCode=0x400d1fa8 <blink_task>) at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/freertos/include/freertos/task.h:427
  6    Thread 1073412716 (esp_timer) xTaskCreate (pvCreatedTask=0x0, uxPriority=5, pvParameters=0x0, usStackDepth=768, pcName=0x3f402ba8 "blink_task", 
    pvTaskCode=0x400d1fa8 <blink_task>) at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/freertos/include/freertos/task.h:427
  5    Thread 1073426924 (ipc1) xTaskCreate (pvCreatedTask=0x0, uxPriority=5, pvParameters=0x0, usStackDepth=768, pcName=0x3f402ba8 "blink_task", 
    pvTaskCode=0x400d1fa8 <blink_task>) at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/freertos/include/freertos/task.h:427
  4    Thread 1073438080 (Tmr Svc) xTaskCreate (pvCreatedTask=0x0, uxPriority=5, pvParameters=0x0, usStackDepth=768, pcName=0x3f402ba8 "blink_task", 
    pvTaskCode=0x400d1fa8 <blink_task>) at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/freertos/include/freertos/task.h:427
  3    Thread 1073434036 (IDLE) xTaskCreate (pvCreatedTask=0x0, uxPriority=5, pvParameters=0x0, usStackDepth=768, pcName=0x3f402ba8 "blink_task", 
    pvTaskCode=0x400d1fa8 <blink_task>) at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/freertos/include/freertos/task.h:427
  2    Thread 1073435424 (IDLE : Running @CPU1) 0x400d1afe in esp_vApplicationIdleHook ()
    at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/components/esp32/./freertos_hooks.c:62
* 1    Thread 1073432516 (main : Running @CPU0) app_main ()
    at /Users/kangrich/Documents/tools/esp-idf-v3.0.5/examples/get-started/blink/main/./blink.c:43
```

igrr, 2018-11-05:
@RA5040 @kangks New release has been published at https://github.com/espressif/openocd-esp32/releases/, could you please give it a try?

zhivko, 2018-11-21:
I have similar issue:
```
c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd.exe -v
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Restore debug stubs @ 0 on core0 of target 'esp32'

c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Detected debug stubs @ 3ffb3b50 on core0 of target 'esp32'
Info : xtensa_poll: Target offline
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 700ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 1500ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 3100ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 6300ms
Info : accepting 'gdb' connection on tcp/3333

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 6300ms
```

gerekon, 2018-11-22:
@zhivko It seems to be another issue. Could you open separate topic for it to keep threads a bit organized? Also please provide full info regarding your problem as it us done in the beginning of this thread.

zhivko, 2018-11-22:
@gerekon created https://github.com/espressif/openocd-esp32/issues/64

igrr, 2018-11-22:
@RA5040 @kangks Closing this issue as new release has been published, please feel free to reopen if the issue persists.

closed 2018-11-22 by igrr