espressif/openocd-esp32 #60: Cannot debug larger programs

RA5040, 2018-09-07:
I am running openocd on a ESP32 Devkit C. The debugging is fine on a small program with a single thread. When I try to run it on a larger program with 6 thread (+ main) it fails with the following openocd log. 

[openocdlog.zip](https://github.com/espressif/openocd-esp32/files/2362356/openocdlog.zip)

Also the jtag link on ttyUSB1 is disconnected, as I show.

RA5040, 2018-09-08:
Having spent a considerable amount of time trying to figure out what is wrong, I eventually found out that using GPIO pin 32 on the Wrover Kit doesn't work and neither does pin 33.That may be down to me not configuring the board correctly ... but as the 'configuration' requires desoldering and resoldering tiny surface-mount resistors I think I'll pass for now.

At any rate, that problem is now 'resolved', so the issue is not with program size.

However here is another example of gdb failing on Eclipse:
			nk.resetDisplay();
	bpt1		digitalWrite(beeperPin, 1);
			vTaskDelay(200 / portTICK_PERIOD_MS);
			digitalWrite(beeperPin, 0);
			nk.displayError(atemp, fireError);
	bpt2		vTaskDelay(200 / portTICK_PERIOD_MS);

The debug will break at bpt1 and I can Continue to bpt2 and Continue again and it will break again at bpt1 ... no problem.

However if I single-step from bpt1 or bpt2 OpenOCD crashes out.
The code is in a task ... and this seems to happen whenever I do a Step-Over or Step-Into when the break is in a task (but I haven't tried this exhaustively).

Single-stepping in app-main seems to be fine.

I have spent ages trying to get debugging to work properly on the Wrover Kit and instead of debugging my applications, I'm debugging gdb and openocd and the Wrover kit.  It's an utter waste of time, and in my opinion the product should be taken off the market by Espressif, as they make the claim in their documentation that it is perfect for hardware debugging (I will be looking for money back from them ... good luck to me!).

I would be interested to know if anyone has a similar experience with other JTAG debuggers.  Perhaps the fault is with OpenOCD and not the Wrover Kit?

RA5040, 2018-09-08:
BTW, this is not just in Eclipse. Here is the gdb log from the command line:
874				nk.resetDisplay();
875				digitalWrite(beeperPin, 1);
876				vTaskDelay(200 / portTICK_PERIOD_MS);
(gdb) n
Target halted. PRO_CPU: PC=0x4000BFF0             APP_CPU: PC=0x400D2B54 (active)
/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) y

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.

/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,

I guess I need to report this to gnu.  But perhaps someone here has some workarounds/suggestions?

igrr, 2018-09-09:
Hi @RA5040,

GPIOs 32 and 33 are connected to a 32kHz XTAL on Wrover-kit, so can't be used as GPIOs.

With regards to GDB assertions, this is not specific to WROVER-KIT, it is a GDB issue, which happens when a the compiler inlines a function into the task function (i.e. into the outer stack frame). It is tracked here: https://github.com/espressif/esp-idf/issues/2343

We will have a workaround for this issue in IDF soon, and a fix in new versions of GDB released after a month or so.

RA5040, 2018-09-09:
Thanks @igrr 
Is there anywhere to see a summary of current serious problems (like this one), with a workaround if applicable and a fix schedule?  At the moment what I'm doing is stumbling on bugs and spending a lot of time (with help from you guys, for sure!) figuring out if it's my fault or not and what to do about it. It's a terrible (but avoidable) waste of time and energy.

Regarding the Wrover-kit ... well, that's what Espressif have decided to do, no doubt in their wisdom (although the point of having a pin marked 'GPIOx' which can't be used as a GPIO eludes me).

Regarding the in-line issue. What I really want to do is to disable all inline expansions for debugging. The Og switch obviously doesn't do that, so what's needed is an O0 switch. Is that what you are working on?  Is there a way of doing it manually?

If not, is there a way of passing '-fno-inline-small-functions' to gcc? 

If not, does '__attribute__ ((noinline))' work?  I'll try it, course, but it will only work with my code, not with library routines ... so I guess these need to be 'Continued' over?

Do you have an estimate of when this will be fixed?

closed 2018-09-10 by RA5040