espressif/openocd-esp32 #61: light sleep causes openOCD to lose CPU

crashdox, 2018-09-27:
see https://www.esp32.com/viewtopic.php?f=13&t=7451

>> Your observation is correct, when the CPU is clock gated (in light sleep) or powered down (in deep sleep), OpenOCD sees that as CPU being lost... We might be able to do something about light sleep, I think. Please open an issue in openocd-esp32 repository if this is important for you, so that we can track this request.

light sleep supporting debug would be great.

NeilMacMullen, 2019-02-27:
I would also like that as a feature.  Power management is a major feature in our application so it sleeps almost immediately after boot!  That means the only practical way to use jtag for upload of a new application is to force a boot into the bootloader which is inconvenient.

erhankur, 2023-01-17:
Closed due to inactivity.

closed 2023-01-17 by erhankur