espressif/openocd-esp32 #62: internal-error: void inline_frame_this_id: Assertion !frame_id_eq

ivankravets, 2018-10-18:
Hi,

We receive a lot of bug reports with the same problem from our customers last time. What is interesting, that the latest release works well with Arduino but does not work with ESP-IDF. 

# Log

## OpenOCD

```
/Users/ikravets/.platformio/packages/tool-openocd-esp32/bin/openocd -s /Users/ikravets/.platformio/packages/tool-openocd-esp32/share/openocd/scripts -f interface/ftdi/olimex-arm-usb-tiny-h.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev (2018-09-20-12:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
esp32 interrupt mask on
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x400E3ABE (active)    APP_CPU: PC=0x400E3ABE
Info : Detected debug stubs @ 3ffc2d70 on core0 of target 'esp32'
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : accepting 'gdb' connection on tcp/3333
Info : Target halted. PRO_CPU: PC=0x4009175E (active)    APP_CPU: PC=0x400E3ABE
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 79 KB
Info : Target halted. PRO_CPU: PC=0x4009175E (active)    APP_CPU: PC=0x400E3ABE
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Target halted. PRO_CPU: PC=0x4009175E (active)    APP_CPU: PC=0x400E3ABE
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 79 KB
Info : Using flash size 80 KB
Info : Target halted. PRO_CPU: PC=0x4009175E (active)    APP_CPU: PC=0x400E3ABE
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 79 KB
Info : Using flash size 28 KB
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
Info : Target halted. PRO_CPU: PC=0x400D08D0 (active)    APP_CPU: PC=0x400E3ABE
Info : Detected debug stubs @ 3ffc2d70 on core0 of target 'esp32'
```

## GDB

```
/Users/ikravets/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gdb /Volumes/SOFT/Projects/Misc/platformio_test_projects/debug/esp32-espidf/.pioenvs/esp32dev/firmware.elf
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_apple-darwin16.3.0 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Volumes/SOFT/Projects/Misc/platformio_test_projects/debug/esp32-espidf/.pioenvs/esp32dev/firmware.elf...done.
(gdb) target remote :3333
Remote debugging using :3333
0x400e3abe in esp_vApplicationWaitiHook ()
    at /Users/ikravets/.platformio/packages/framework-espidf/components/esp32/freertos_hooks.c:66
66	    asm("waiti 0");
(gdb) mon reset halt
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
(gdb) flushregs
Register cache flushed.
(gdb) tb app_main
Temporary breakpoint 1 at 0x400d08d0: file src/main.c, line 28.
(gdb) c
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
Target halted. PRO_CPU: PC=0x400D08D0 (active)    APP_CPU: PC=0x400E3ABE
[New Thread 1073510484]
[New Thread 1073508584]
[New Thread 1073513140]
[New Thread 1073500792]
[New Thread 1073497508]
[New Thread 1073499320]
[New Thread 1073501924]
[Switching to Thread 1073506468]

Temporary breakpoint 1, app_main () at src/main.c:28
28	{
(gdb) i thread
  Id   Target Id         Frame
  8    Thread 1073501924 (dport) 0x40084c22 in vTaskDelete (xTaskToDelete=<optimized out>)
    at /Users/ikravets/.platformio/packages/framework-espidf/components/freertos/tasks.c:1316
  7    Thread 1073499320 (ipc0) 0x40084184 in xQueueGenericReceive (xQueue=0x3ffc4860, pvBuffer=0x0,
    xTicksToWait=2169353893, xJustPeeking=0)
    at /Users/ikravets/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  6    Thread 1073497508 (esp_timer) 0x40084184 in xQueueGenericReceive (xQueue=0x3ffc354c, pvBuffer=0x0,
    xTicksToWait=2169353893, xJustPeeking=0)
    at /Users/ikravets/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  5    Thread 1073500792 (ipc1) 0x40084184 in xQueueGenericReceive (xQueue=0x3ffc4e20, pvBuffer=0x0,
    xTicksToWait=2169353893, xJustPeeking=0)
    at /Users/ikravets/.platformio/packages/framework-espidf/components/freertos/queue.c:1591
  4    Thread 1073513140 (Tmr Svc)
/Volumes/build/idf/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: void inline_frame_this_id(struct frame_info *, void **, struct frame_id *): Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n)
```

## Firmware

See attachmets.

[firmware.elf.zip](https://github.com/espressif/openocd-esp32/files/2492820/firmware.elf.zip)

mahavirj, 2018-10-21:
@ivankravets Also reported on ESP-IDF github issues page, solution along with timelines can be found at https://github.com/espressif/esp-idf/issues/2343#issuecomment-417853341

Relevant commit that should fix this issue https://github.com/espressif/esp-idf/commit/3ba63a520c91b961a2632ad7385c8bac9039e197 (This will be part of next stable ESP-IDF release v3.2)

ivankravets, 2018-10-22:
We did a quick patch on our side. Now, debugging should work. Please open [PlatformIO IDE](https://platformio.org/platformio-ide) Terminal and type
```
pio update
pio upgrade --dev
```

closed 2018-10-22 by ivankravets

igrr, 2019-01-05:
The fix was in IDF, it was to prevent having inline functions in outer stack frames. If you

nospam2000, 2019-01-05:
> The fix was in IDF, it was to prevent having inline functions in outer stack frames

Thanks! My file `~/michael/.platformio/packages/framework-espidf/components/freertos/port.c` is from November 1st and doesn't contain the new code after updating using the commands `pio update ; pio upgrade --dev`
```
Platform Espressif 32
--------
Updating espressif32                     @ 1.5.0          [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10000.0      [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
Updating framework-espidf                @ 3.30101.0      [Up-to-date]
Updating tool-esptoolpy                  @ 1.20500.0      [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ febfe90        [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10000.0      [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
Updating framework-espidf                @ 3.30101.0      [Up-to-date]
Updating tool-esptoolpy                  @ 1.20500.0      [Up-to-date]```
```

That fix depends on setting `CONFIG_FREERTOS_TASK_FUNCTION_WRAPPER` to 1, which means
the sdkconfig.h has to be adapted after the new version is downloaded, I don't know if this happens automagically.

**In the meanwhile I discovered my own solution: disable the inline functions at all** in the build settings of `platformio.ini`:
`build_flags = -g -Og -fno-inline` 

The most important switch is `-fno-inline`. With this configuration debugging works for me.

 Michael

ivankravets, 2019-01-09:
The latest release of ESP-IDF which we support now is v3.1.1. We plan to add support for 3.1.2 soon.