espressif/openocd-esp32 #64: Issue with running gdb on esp32 Wrover Dev Kit

zhivko, 2018-11-22:
Environment
Development Kit: ESP32-Wrover-Kit

esp-idf-v release/v3.2
openocd-esp32-win32-0.10.0-esp32-20181105.zip

Development Env:	Command line

Operating System:	Windows 10 64bit

Problem Description
OpenOcd not started correctly - not possible to connect to it with xtensa-esp32-elf-gdb

Make and flash the program
Run OpenOCD
Run xtensa-esp32-elf-gdb
You get:
```
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 6300ms
```

Expected Behavior
Flash the program, then run OpenOCD, then run the xtensa gdb and hardware debugging should work.
I.e. xtensa-esp32-elf-gdb should be able to connect to openocd.

Actual Behavior
xtensa-esp32-elf-gdb could not connect

Steps to reproduce
Using a ESP32WroverKit with built-in JTAG and JTAG jumpers in place as in https://docs.espressif.com/projects/esp-idf/en/latest/get-started/get-started-wrover-kit.html, follow the Espressif instructions to debug with JTAG in Windows to the letter, build and flash one of the IDF examples (for example blink), and you should see that dbg does not work.

zhivko, 2018-11-22:
HEre is output from console:
```
c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd.exe -v
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Restore debug stubs @ 0 on core0 of target 'esp32'

c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Detected debug stubs @ 3ffb3b50 on core0 of target 'esp32'
Info : xtensa_poll: Target offline
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 700ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 1500ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 3100ms
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 6300ms
Info : accepting 'gdb' connection on tcp/3333

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Polling target esp32 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 6300ms
```
Attached -d3 debug output.
[out.txt](https://github.com/espressif/openocd-esp32/files/2605774/out.txt)

using board/**esp32-wrover.cfg**:
```
c:\msys32\home\klemen\esp\openocd-esp32>.\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover.cfg -d3
```
I also could not debug - output is here: 
[out2.txt](https://github.com/espressif/openocd-esp32/files/2605823/out2.txt)

igrr, 2018-11-22:
Which program is running in the ESP32 when you are trying this? It seems like the target responds initially but then stops responding, which might indicate that the program running on the ESP32 reconfigures JTAG pins for some other purpose. Could you please try flashing "blink" example from ESP-IDF and then starting OpenOCD?

> using board/esp32-wrover.cfg:

Regarding the config file, you should use esp32-wrover.cfg if your WROVER-KIT board has a WROVER module. Some early WROVER-KIT boards have been produced with WROOM module, for these `esp-wroom-32.cfg` should be used.

gerekon, 2018-11-22:
@zhivko Could you also provide detailed log files as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-reporting-issues)?

zhivko, 2018-11-22:
@igrr yes you are right I havent flashed blink firmware while doing that test. It could be that pins are used for other functions. Will retest with blink firmware and report how it goes.

gerekon, 2018-11-27:
@zhivko Does you problem still exist? Can you debug blink example? I do not see critical problems in log.
If you have problems with debugging blink example could you also provide detailed log files as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/tips-and-quirks.html#jtag-debugging-tip-reporting-issues)?

zhivko, 2018-11-27:
Kind a works now, but I cannot step through code - it seems eclipse connect succesfully to openocd.
Any idea what I need to set to see code?
[image]

gerekon, 2018-11-27:
@zhivko What app do you debug? According to your log you write `hello_world` binary and try to stop in `blink.cpp`. That seemed confusing. Could you try on pure 'blink' example from ESP-IDF?

zhivko, 2018-11-28:
It is same problem with blink app.
In eclipse debuger console I get:
```
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-host_pc-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".

No line 43 in file "C:\\msys32\\home\\klemen\\esp\\blink\\main\\blink.c".
```

zhivko, 2018-11-28:
Is it possible that double backslash is not correctly understood by eclipse or reported from xtensa-esp32-elf-gdb ? Reported as double backslash instead of single one? Do I need to specify source lookup in eclipse somwhere?

zhivko, 2018-11-28:
If I enter info sources in console of xtensass-esp32-elf-gdb.exe I dont get blink.c to be included. I get:

```
info sources
Source files for which symbols have been read in:

Source files for which symbols will be read in on demand:

 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fputs.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/errno/../../../.././newlib/libc/errno/errno.c
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\esp_timer.c
 /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c
 /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\include\soc\dport_access.h
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\rtc_init.c
 C:\msys32\home\klemen\esp\esp-idf\components\newlib\time.c
 /Users/igrokhotkov/e/esp32/hal/hal/state_asm.S
 /Users/igrokhotkov/e/esp32/hal/hal/interrupts.c
 /Users/igrokhotkov/e/esp32/hal/hal/windowspill_asm.S
 /Users/igrokhotkov/e/esp32/hal/hal/clock.S
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\include\rom\uart.h
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\clk.c
 C:\msys32\home\klemen\esp\esp-idf\components\driver\periph_ctrl.c
 C:\msys32\home\klemen\esp\esp-idf\components\driver\timer.c
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\include\freertos\portable.h
 C:\msys32\home\klemen\esp\esp-idf\components\spi_flash\cache_utils.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\src\memory_layout_utils.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\soc_memory_layout.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\cpu_util.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\rtc_wdt.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\rtc_time.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\rtc_clk_common.h
 C:\msys32\home\klemen\esp\esp-idf\components\soc\esp32\rtc_clk.c
 C:\msys32\home\klemen\esp\esp-idf\components\newlib\reent_init.c
 C:\msys32\home\klemen\esp\esp-idf\components\newlib\syscalls.c
 C:\msys32\home\klemen\esp\esp-idf\components\newlib\locks.c
 /tmp/build/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libm/common/../../../.././newlib/libm/common/s_fpclassify.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/string/../../../.././newlib/libc/string/u_strerr.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/string/../../../.././newlib/libc/string/strerror.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libm/math/../../../.././newlib/libm/math/s_frexp.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/mprec.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/dtoa.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/string/../../../.././newlib/libc/string/xpg_strerror_r.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/sprintf.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/snprintf.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/reent/../../../.././newlib/libc/reent/reent.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/puts.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/printf.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c
 /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c
 C:\msys32\home\klemen\esp\esp-idf\components\soc\include\soc\soc_memory_layout.h
 C:\msys32\home\klemen\esp\esp-idf\components\log\log.c
 C:\msys32\home\klemen\esp\esp-idf\components\heap\heap_caps_init.c
 C:\msys32\home\klemen\esp\esp-idf\components\heap\multi_heap_platform.h
 C:\msys32\home\klemen\esp\esp-idf\components\heap\multi_heap.c
 C:\msys32\home\klemen\esp\esp-idf\components\heap\heap_private.h
 C:\msys32\home\klemen\esp\esp-idf\components\heap\heap_caps.c
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\xtensa_vector_defaults.S
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\timers.c
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\xtensa_intr_asm.S
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\xtensa_context.S
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\xtensa_vectors.S
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\portasm.S
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\xtensa_intr.c
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\include\freertos\portmacro.h
 C:\msys32\home\klemen\esp\esp-idf\components\freertos\queue.c
[... cut ...]
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\int_wdt.c
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\dport_panic_highint_hdl.S
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\include\esp_panic.h
 C:\msys32\home\klemen\esp\esp-idf\components\esp32\panic.c
 c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h
 C:\msys32\home\klemen\esp\esp-idf\components\cxx\cxx_guards.cpp
 C:\msys32\home\klemen\esp\esp-idf\components\cxx\cxx_exception_stubs.cpp
```

zhivko, 2018-11-29:
To illustrate issue here is recorded video.
https://youtu.be/fvCdl0diQmo

gerekon, 2018-11-29:
@zhivko Have yoou programmed bootloader and partition table into flash? Could you do it using `make flash` command and try again?

zhivko, 2018-11-29:
Yes of course - I also tried that and without
```
mon program_esp32 c:/msys32/home/klemen/esp/DoubleLifter/build/DoubleLifter.bin 0x10000 verify
```
doesn't help

gerekon, 2018-11-29:
I have no problem with your config and blink example.
I use Eclipse:
- Version: Oxygen.3a Release (4.7.3a)
- Build id: 20180405-1200

Could you try my binaries?
[blink.zip](https://github.com/espressif/openocd-esp32/files/2628912/blink.zip)

Does blink example run normally without debugger after flashing it with 'make flash'? It should blink with RGB LED or LCD screen depending on example GPIO pin config.
Could you make a photo of your ESP32-Wrover-Kit with HW connections and jumpers on it?

gerekon, 2018-11-29:
Could you send me your blink project?

zhivko, 2018-11-29:
[blink.zip](https://github.com/espressif/openocd-esp32/files/2628974/blink.zip)

zhivko, 2018-11-29:
Your example works - green led is flashing - what pin number is it? I will
use same in my example to see if my example is working the same.

gerekon, 2018-11-29:
Hmm, HW connections seem to be OK.

> Your example works - green led is flashing - what pin number is it?

Pin number 2

gerekon, 2018-11-29:
Can we set up remote debug session? For example via TeamViewer?

zhivko, 2018-11-29:
Of course at 17:30 CET?

gerekon, 2018-11-29:
> Of course at 17:30 CET?

Ok. Send me the session ID and pin code to alexey at espressif.com

zhivko, 2018-11-30:
Yesterday we had teamviewer session with Alexey, and we found out that if firmware is prepared via msys2 compilation, than it is possible to step through code.
On other hand if it is prepared via eclipse command:
python ${IDF_PATH}/tools/windows/eclipse_make.py -j6
Visible at:
[image]
Debugging don't work in a following sense:
1) code execution does not stop at breakpoints
2) source file is not opened instead "No source file named ..." is visible at Eclipse debugger console

gerekon, 2018-11-30:
Actually the problem in the image built using `python ${IDF_PATH}/tools/windows/eclipse_make.py -j6` from Eclipse. It causes board to enter infinite reset loop so debugging does not work. It seems not to be issue of the debugger. 
When image is build using msys command line everything works.

igrr, 2018-11-30:
@zhivko Could you please upload the .elf file produced by eclipse_make.py (the one which results in infinite reset loop)?

zhivko, 2018-12-02:
Here it is.
[blink.zip](https://github.com/espressif/openocd-esp32/files/2637335/blink.zip)

zhivko, 2018-12-20:
Anything new on this topic?

WayneKeenan, 2019-09-28:
I was also seeing the Core/Debug reset messages after switching from one type of dev board to another.

I had to alter:  MenuConfig -> Components -> ESP-32 Specific -> Main XTAL frequency.  

I changed from from "Autodetect" to a specific frequency (in my case is was 40 Mhz)

sdkconfig snippet
```
CONFIG_ESP32_XTAL_FREQ_40=y
CONFIG_ESP32_XTAL_FREQ_26=
CONFIG_ESP32_XTAL_FREQ_AUTO=
CONFIG_ESP32_XTAL_FREQ=40
```

HTH

rebeliousconformist, 2020-01-08:
+1 having the same issue 

details of my current setup are as https://github.com/espressif/esp-idf/issues/4601 with compiler esp-2019r2

erhankur, 2023-01-17:
Closed due to inactivity.

closed 2023-01-17 by erhankur