espressif/openocd-esp32 #65: Cannot upload program using OpenOCD

yaqwsx, 2018-11-22:
I have ESP32 DevkitC with ESP-Prog wired as shown in [2]. When I try to upload a program using OpenOCD (`bin/openocd -s scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 firmware.bin 0x10000 verify exit"`), the verification phase fails. The log can be found at [1].

I think all my wiring and invoking of OpenOCD is correct, could be an issue with OpenOCD?

[1] https://pastebin.com/JXXL2RkJ
[2] https://photos.app.goo.gl/S3WQX1NpRbjddPXV7

sinope-db, 2018-11-22:
I see you're running at 20MHz, have you tried reducing adapter speed (10 or 3 MHz)?
https://github.com/espressif/openocd-esp32/blob/05cc1857bfbd9ca44af94a31e461f77161b5af3b/tcl/board/esp-wroom-32.cfg#L37

igrr, 2018-11-23:
Hi @yaqwsx, thanks for the report. I've replicated your setup and was able to flash and verify successfully. The difference in the log i see is that in your case there is

```
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Warn : Empty flash mapping!
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
```

while in my case i see

```
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400 
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 18 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 72 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400 
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
```

Could you please add `-d3` flag to OpenOCD command line and share the logs with this flag?

yaqwsx, 2018-11-23:
I tried to add the `-d3` flag and then the programming magically succeded (log [2]) and the firmware flashing succeeded as the debug logs slowed down the communication (log [2]). Therefore, I lowered the communication frequency to 200 kHz and I was able to flash a firmware (log [1]). However, lowering the frequency does not solve the problem - the interface gets so slow, that debugger frontend times outs.

First, I thought the problem might be caused by a poor wiring and the signal could be deformed (haven't verified on a scope yet), however, and correct me if I am wrong, the debug messages introduce only a delay between commands and do not affect the clock speed itself (as it should be generated by the FTDI chip itself), therefore, the wiring shouldn't be the root of the problem.

Is my reasoning correct? Might it be after all a problem with OpenOCD?

EDIT: I tried it once more (log [3]) and if fail again. I will look at the signals with a scope, if you find something interesting in the logs, please let me know.

[1] https://pastebin.com/g7F3DSUc
[2] https://pastebin.com/zDAcnQwe
[3] https://pastebin.com/ePvyQmxn

yaqwsx, 2018-11-25:
I tried to capture the waveform using my scope. Yellow is clock, blue is TMS and red is TDI. The waveform for 2 MHz and 200 kHz looks fine - even there are some over-, and under-shots I don't think they should cause a problem. The 200 Mhz waveform looks scary, however I think I am reaching limits of my 70Mhz scope here. All the examples, no matter what frequency I used, failed.

200 kHz:
[image]

2 MHz:
[image]

20 MHz:
[image]

yaqwsx, 2018-12-27:
I haven't been successful in debugging ESP32 yet. Do you have any ideas what to try? Do the waveform look good?

yaqwsx, 2019-01-03:
After playing for a while I have a conclusion: OpenOCD preserves original content of the flash.

yaqwsx, 2019-05-02:
I tried to replicate the setup on a new computer with a fresh OS installation and the problem preserves. 

Is the Czech Espressif office active? Would it be possible to arrange a meeting and discuss the problem in person so the actual hardware could be checked?

gerekon, 2019-05-03:
@yaqwsx Did you tried the last version of OpenOCD?

gerekon, 2019-05-03:
Also as I see you power the devkitC from ESP-prog board. Could you try to power it using its own USB port and separate cable?

notjulian, 2019-07-17:
hi @yaqwsx ,
try to load for the first time all the partition
`c:\ESP\openocd-esp32\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 C:/ESP/build/ota_data_initial.bin 0xd000 verify exit"
c:\ESP\openocd-esp32\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 C:/ESP/build/bootloader/bootloader.bin 0x1000 verify exit"
c:\ESP\openocd-esp32\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 C:/ESP/emitter_client.bin 0x10000 verify exit"
c:\ESP\openocd-esp32\bin\openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 C:/ESP/partitions_two_ota.bin 0x8000 verify exit"
`

not only
`bin/openocd -s scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp32 firmware.bin 0x10000 verify exit"`

pridgeway-rsc, 2020-10-19:
I was having this problem earlier today when I upgrade my OpenOCD_esp32 tool and found a fix for it.

The root problem is in file **/openocd-esp32/share/openocd/scripts/target/esp32.cfg** with how the **program_esp32** function calls **program_esp** when these files were refactor to move some of the argument logic to **/openocd-esp32/share/openocd/scripts/target/esp_common.cfg**.  The way it's written the **args** parameter is being past as a single {list} instead individual arguments.

The fix is to change the following function call in file **/openocd-esp32/share/openocd/scripts/target/esp32.cfg** around line 64 from `program_esp $filename $args` to `program_esp $filename {*}$args`

This will allow the arguments to properly flow to the next function.  I'm working on putting in the pull request right now.

As a side note, ironically you can just change the **program_esp32** call to **program_esp** in your original OpenOCD programming command line: `bin/openocd -s scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp firmware.bin 0x10000 verify exit"` and never would of seen this error.

gerekon, 2021-01-11:
@yaqwsx If it is still actual. Could you check the problem with the latest release (https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20201202)?

yaqwsx, 2021-01-12:
@gerekon: I don't have an ESP32 laying around right now, I will try to test it within a week and let you know.

closed 2022-09-01 by gerekon