espressif/openocd-esp32 #66: does it support ESP32-SOLO-1?

tatsutaigu, 2018-12-19:
Hi,

I would like to debug ESP32-SOLO-1.
I found below error.
And, I cannot use gdb.

> $ /home/foo/esp/openocd-esp32/bin/openocd.exe -s /home/foo/esp/openocd-esp32/share/openocd/scripts -f interface/ftdi/um232h.cfg  -c 'set ESP32_ONLYCPU 1' -f board/esp-wroom-32.cfg
> Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> 0
> adapter speed: 20000 kHz
> esp32 interrupt mask on
> Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
> Info : clock speed 20000 kHz
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
> Warn : JTAG tap: esp32.cpu1       UNEXPECTED: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
> Error: JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Error: Trying to use configured scan chain anyway...
> Error: esp32.cpu1: IR capture error; saw 0x1f not 0x01
> Warn : Bypassing JTAG setup events due to errors
> Info : accepting 'gdb' connection on tcp/3333
> Error: Target not examined yet
> 
> 
> Error: Target not halted
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected
> Info : accepting 'gdb' connection on tcp/3333
> Error: Target not examined yet

> $ xtensa-esp32-elf-gdb -ex 'target remote localhost:3333' ./build/blink.elf
> GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 7.10
> Copyright (C) 2015 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "--host=i686-host_pc-mingw32 --target=xtensa-esp32-elf".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from ./build/blink.elf...done.
> Remote debugging using localhost:3333
> Remote connection closed
> (gdb)
>

igrr, 2019-02-03:
Support for single core version of the chip has been added in https://github.com/espressif/openocd-esp32/commit/c97000647ebfcada43a550730b477147c621c617. Documentation update of IDF programming guide is still pending.

igrr, 2019-02-13:
Has been relased in https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20190212. You can use `-f boards/esp32-solo-1.cfg` instead of `-c 'set ESP32_ONLYCPU 1' -f board/esp-wroom-32.cfg`.

closed 2019-02-13 by igrr