espressif/openocd-esp32 #70: Using openI flashed the blink example using ocd with a buspirate 3.6  to debug a esp-wroom32

jaoh, 2019-02-03:
I flashed the link example using make flash.  

I want to use jtag and oocd to debug my esp . I get the following output

sudo openocd -s share/openocd/scripts -f interface/buspirate.cfg -f board/esp-wroom-32.cfg 
Open On-Chip Debugger 0.10.0-dev-00009-g049f90de (2019-02-03-14:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'buspirate' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
Warn : Transport "jtag" was already selected
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
Info : Buspirate Interface ready!
Info : This adapter doesn't support configurable speed
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x5F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x5F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
^CInfo : Shutting down buspirate.
Info : Buspirate switched to normal mode

I use the following interface configuration

#
# Buspirate with OpenOCD support
#
# http://dangerousprototypes.com/bus-pirate-manual/
#

interface buspirate

# you need to specify port on which BP lives
buspirate_port /dev/ttyUSB0

# communication speed setting
buspirate_speed normal ;# or fast

# voltage regulator Enabled = 1 Disabled = 0
buspirate_vreg 1

# pin mode normal or open-drain
#buspirate_mode normal

# pullup state Enabled = 1 Disabled = 0
#buspirate_pullup 0

# this depends on the cable, you are safe with this option
reset_config srst_only

igrr, 2019-02-03:
"Debug controller 0 was reset" message does not indicate an error; it means that the CPU keeps resetting. You can open UART port using `make monitor`and then observe serial output. It will probably indicate reset reason.

jaoh, 2019-02-04:
My esp currently receives +5V and GND from my JTAG device. to access the UART I would have to use the USB port of my devkit. This would also supply power to the system. Do I have to disconnect +5V and GND form my Jtag device or do I have to connect to the UART in another way?

igrr, 2019-02-04:
You can disconnect 5V line from the JTAG (so that the devkit will receive power from USB) but keep GNDs of devkit and JTAG connected.

jaoh, 2019-02-04:
I tried this 

this is my serial output 

Toolchain path: /home/jaoh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/jaoh/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6232
load:0x40078000,len:11164
load:0x40080400,len:6672
entry 0x40080764
I (29) boot: ESP-IDF v3.3-beta1-328-gabea9e4c0 2nd stage bootloader
I (29) boot: compile time 17:36:39
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08798 ( 34712) map
I (106) esp_image: segment 1: paddr=0x000187c0 vaddr=0x3ffb0000 size=0x01e8c (  7820) load
I (109) esp_image: segment 2: paddr=0x0001a654 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/jaoh/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (114) esp_image: segment 3: paddr=0x0001aa5c vaddr=0x40080400 size=0x055b4 ( 21940) load
I (132) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x12720 ( 75552) map
0x400d0018: _flash_cache_start at ??:?

I (158) esp_image: segment 5: paddr=0x00032740 vaddr=0x400859b4 size=0x025e0 (  9696) load
0x400859b4: vPortCPUReleaseMutexIntsDisabledInternal at /home/jaoh/esp/esp-idf/components/freertos/portmux_impl.inc.h:157
 (inlined by) vPortCPUReleaseMutexIntsDisabled at /home/jaoh/esp/esp-idf/components/freertos/portmux_impl.h:109
 (inlined by) vPortCPUReleaseMutex at /home/jaoh/esp/esp-idf/components/freertos/port.c:421

I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (170) cpu_start: Pro cpu up.
I (174) cpu_start: Application information:
I (179) cpu_start: Project name:     blink
I (183) cpu_start: App version:      1
I (188) cpu_start: Compile time:     17:36:42
I (193) cpu_start: Compile date:     Feb  3 2019
I (198) cpu_start: ESP-IDF:          v3.3-beta1-328-gabea9e4c0
I (205) cpu_start: Starting app cpu, entry point is 0x40080f70
0x40080f70: call_start_cpu1 at /home/jaoh/esp/esp-idf/components/esp32/cpu_start.c:265

I (0) cpu_start: App cpu up.
I (215) heap_init: Initializing. RAM available for dynamic allocation:
I (222) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (228) heap_init: At 3FFB2ED0 len 0002D130 (180 KiB): DRAM
I (234) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (241) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (247) heap_init: At 40087F94 len 0001806C (96 KiB): IRAM
I (253) cpu_start: Pro cpu start user code
I (271) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 344 lines cut ...]
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...done.
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Ignoring packet error, continuing...
gdbinit:1: Error in sourced command file:
Bogus trace status reply from target: timeout
(gdb) ^CQuit
(gdb) quit

this is my gdbinit file 

:~/esp/blink$ cat gdbinit 
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c

One thing I noticed is that the frequency of the resets is much lower if use the sserial connection. 
could it be possible that my jtag connection is wrong ?

gerekon, 2021-01-11:
@jaoh Could you check if problem still exists with the latest OpenOCD release (https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20201202)?

closed 2022-09-01 by gerekon