espressif/openocd-esp32 #72: Error when attempting flash download using Segger Jlink Base to Esp32 (16MB)

NeilMacMullen, 2019-02-27:
We have manufactured a board using an ESP32-Wroom module (16MB flash).   I'm using the latest OpenOCD-esp32 tools: **"Open On-Chip Debugger 0.10.0-dev (2019-02-12-02:52)"** 

Issuing this command...
`
.\bin\openocd.exe -d 3 -s share/openocd/scripts -f interface/jlink.cfg -f board/esp-wroom-32.cfg -c "program_esp32 firmware.bin 0x10000 verify exit"`

I see that the cores are reset but OpenOCD is unable to write the 4 bytes for the first watchdog area. (This occurs regardless of whether i boot the chip into the bootloader or the application).

The full log is included below but the relevant portion is.. 

> Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Error: cpu0: esp32_fetch_all_regs (line 227): DSR (00000088) indicates DIR instruction generated an overrun!
> Error: cpu1: esp32_fetch_all_regs (line 227): DSR (00000088) indicates DIR instruction generated an overrun!
> Info : Target halted. PRO_CPU: PC=0x40062233 (active)    APP_CPU: PC=0x00000000
> Error: cpu0: xtensa_write_memory (line 761): DSR (8080CC13) indicates DIR instruction generated an exception!
> Warn : esp32: Failed writing 4 bytes at address 0x3FF5F064, data - a1, 3a, d8, 50, a1, 3a, d8, 50
> embedded:startup.tcl:21: Error:

**Full Log (at -d 3)** 

> K:\openocd-esp32>.\bin\openocd.exe -d 3 -s share/openocd/scripts -f interface/jlink.cfg -f board/esp-wroom-32.cfg -c "program_esp32 firmware.bin 0x10000 verify exit"
> Open On-Chip Debugger 0.10.0-dev (2019-02-12-02:52)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> User : 13 4 command.c:544 command_print(): debug_level: 3
> Debug: 14 10 configuration.c:42 add_script_search_dir(): adding share/openocd/scripts
> Debug: 15 20 options.c:96 add_default_dirs(): bindir=/builds/idf/openocd-esp32/openocd-esp32/bin
> Debug: 16 30 options.c:97 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/openocd-esp32/share/openocd
> Debug: 17 42 options.c:98 add_default_dirs(): run_prefix=K:/openocd-esp32/bin
> Debug: 18 50 configuration.c:42 add_script_search_dir(): adding C:\Users\Neil\AppData\Roaming/OpenOCD
> Debug: 19 60 configuration.c:42 add_script_search_dir(): adding K:/openocd-esp32/bin/builds/idf/openocd-esp32/openocd-esp32/share/openocd/site
> Debug: 20 76 configuration.c:42 add_script_search_dir(): adding K:/openocd-esp32/bin/builds/idf/openocd-esp32/openocd-esp32/share/openocd/scripts
> Debug: 21 98 configuration.c:82 find_file(): found share/openocd/scripts/interface/jlink.cfg
> Debug: 22 110 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface jlink
> Debug: 23 122 command.c:143 script_debug(): command - interface ocd_interface jlink
> Debug: 25 130 command.c:323 command_new(): BUG: command 'jlink' does not have the '.usage' field filled out
> Debug: 26 142 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 27 152 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 28 160 command.c:323 command_new(): BUG: command 'jlink freemem' does not have the '.usage' field filled out
> Debug: 29 172 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 30 182 command.c:323 command_new(): BUG: command 'jlink hwstatus' does not have the '.usage' field filled out
> Debug: 31 194 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 32 204 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 33 212 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 34 222 command.c:323 command_new(): BUG: command 'jlink config' does not have the '.usage' field filled out
> Debug: 35 234 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 36 242 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 37 252 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 38 260 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 39 268 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 40 276 command.c:323 command_new(): BUG: command 'config reset' does not have the '.usage' field filled out
> Debug: 41 290 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 42 300 command.c:323 command_new(): BUG: command 'config write' does not have the '.usage' field filled out
> Debug: 43 314 command.c:364 register_command_handler(): registering 'ocd_jlink'...
> Debug: 44 322 configuration.c:82 find_file(): found share/openocd/scripts/board/esp-wroom-32.cfg
> Debug: 45 334 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select jtag
> Debug: 46 346 command.c:143 script_debug(): command - ocd_transport ocd_transport select jtag
> Debug: 47 356 command.c:364 register_command_handler(): registering 'ocd_jtag_flush_queue_sleep'...
> Debug: 48 368 command.c:364 register_command_handler(): registering 'ocd_jtag_rclk'...
> Debug: 49 376 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
> Debug: 50 386 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_assert_width'...
> Debug: 51 398 command.c:364 register_command_handler(): registering 'ocd_scan_chain'...
> Debug: 52 406 command.c:364 register_command_handler(): registering 'ocd_jtag_reset'...
> Debug: 53 416 command.c:364 register_command_handler(): registering 'ocd_runtest'...
> Debug: 54 426 command.c:364 register_command_handler(): registering 'ocd_irscan'...
> Debug: 55 436 command.c:364 register_command_handler(): registering 'ocd_verify_ircapture'...
> Debug: 56 448 command.c:364 register_command_handler(): registering 'ocd_verify_jtag'...
[... 371 lines cut ...]
> in procedure 'program_esp32'
> in procedure 'reset' called at file "share/openocd/scripts/target/esp32.cfg", line 126
> in procedure 'ocd_bouncer'
> in procedure 'ocd_process_reset'
> in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 248
> in procedure 'esp32' called at file "embedded:startup.tcl", line 312
> in procedure 'ocd_bouncer'
> in procedure 'esp_core_halt'
> in procedure 'esp32' called at file "share/openocd/scripts/target/esp32.cfg", line 44
> in procedure 'ocd_bouncer'
> at file "embedded:startup.tcl", line 21
> in procedure 'program_esp32'
> in procedure 'reset' called at file "share/openocd/scripts/target/esp32.cfg", line 126
> in procedure 'ocd_bouncer'
> 
> Debug: 428 10158 command.c:626 run_command(): Command failed with error code -4
> Debug: 429 10168 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_echo ** Unable to reset target **
> Debug: 430 10182 command.c:143 script_debug(): command - echo ocd_echo ** Unable to reset target **
> Debug: 431 10194 log.c:420 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (4254). This may cause trouble with GDB connections.
> User : 434 10216 command.c:762 jim_echo(): ** Unable to reset target **
> Debug: 435 10222 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_shutdown error
> Debug: 436 10236 command.c:143 script_debug(): command - shutdown ocd_shutdown error
> User : 438 10250 server.c:630 handle_shutdown_command(): shutdown command invoked
> Debug: 439 10258 command.c:626 run_command(): Command failed with error code -4
> User : 440 10268 command.c:687 command_run_line():

igrr, 2019-02-28:
In the log I see errors related to communication with your Jlink device:

```
jaylink: Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
jaylink: Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
jaylink: Sending data to device timed out.
jaylink: transport_write() failed: -4.
Error: 326 5862 jlink.c:1772 jlink_flush(): jaylink_jtag_io() failed: JAYLINK_ERR_TIMEOUT.
```

Try lowering `adapter_khz` frequency — it is possible that your Jlink device does not support the frequency requested (20MHz).

NeilMacMullen, 2019-02-28:
Thanks Ivan - this is a really good tip.  Reducing the frequency to 12MHz or below.  There are still a lot of LIBUSB_ERROR_TIMEOUTs.  I suspect this may be because the existing application is using sleep modes.  (There's already an open item for that at https://github.com/espressif/openocd-esp32/issues/61)

For anyone else trying to get this to work, top tips are...
1) In esp-wroom-32.cfg change the adapter_khz value to 12000 or even lower.
2) Try to ensure that you enter the main bootloader before using OpenOCD - i.e. power up the board as if you were going to do a serial update.

Interestingly, the reason I wanted to use jtag for download was because I thought it would be faster than using the serial bootloader.  That turns out to be incorrect; flashing a 1MB application via serial (using 921kbaud) takes around 15 seconds. The same operation using OpenOCD takes around 36 seconds (22 for write, 14 for verify).

igrr, 2019-02-28:
Yes, serial upload uses compression socks it is pretty fast, even if the actual on-the-wire transfer speed is slower than with JTAG. Also with OpenOCD verification is currently done by reading back flash contents, whereas with serial upload the stub on the target calculates a hash of the flash contents and matches it against the one provided by host.

NeilMacMullen, 2019-03-01:
Thanks @igrr   I will close this.

closed 2019-03-01 by NeilMacMullen

hexffffffff, 2020-09-11:
I also see problems when uploading more than 4MB in ESP32. If you use more than 4MB in your module, then it does not start, but reboots all the time. I roughly figured out what the essence of the problem is. The fact ESP32 to download programs through the com port, it is used esptol. But esptol does not load the bootloader directly, it changes the binary "on the fly" by substituting the necessary values into it. But if you use instead of esptool OpenOCD (esp-prog, jlink etc) then it changes nothing in bootloader.bin and writes it "as is".   Because of this, the wrong bootloader gets into the esp32 flash. 

So I see a problem, but I don't know how to fix it. We need some kind of script that will generate bootloader from a binary template, before loading the bootloader. But I don't know how to write it.  Does anyone have any ideas?