espressif/openocd-esp32 #73: Can not start the debugging with OpenOCD and GDB

adexc, 2019-03-15:
Hi,
I'm facing a problem with starting the debug toolchain for ESP32 Thing board using gdb and OpenOCD. After starting gdb i get errors in OpenOCD and GDB displayed, but i don't know what is the problem of the communication between OpenOCD and gdb.

### **Environment:**
- Sparkfun ESP32 Thing
- JTAG Adapter: Olimex ARM-USB-OCD-H
- GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
- ESP-IDF: v3.1.3
- Open On-Chip Debugger 0.10.0-dev (2019-02-12-02:50)
- Development Env: Command line for OCD and GDB / Eclipse 2018-12 (4.10.0) for Build
- Operating System: Linux Mint 19.1 on Virtual Machine

### **Problem Description:**

1. I do following steps:
2. Import the blink project to Eclipse
3. Build and flash the program to ESP32 Thing module with Eclipse and USB
4. Connect the Olimex ARM-USB-OCD-H JTAG Debugger
5. Run OpenOCD 
6. Run xtensa-esp32-elf-gdb

### **Files:**
**gdbinit File**

`set remotelogfile gdblog.txt
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
`

**Starting gdb:**

`$ xtensa-esp32-elf-gdb -x gdbinit build/blink.elf 
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...done.
gdbinit:2: Error in sourced command file:
Remote communication error.  Target disconnected.: Die Verbindung wurde vom Kommunikationspartner zurückgesetzt.`

**gdblog.txt:**
`w +$qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+#c9
r +<Eof>
End of log`

**OpenOCD Logfile:**
`$ bin/openocd -d 3 -s share/openocd/scripts -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f board/esp-wroom-32.cfg 2>&1 | tee openocd.log

Open On-Chip Debugger 0.10.0-dev (2019-02-12-02:50)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 1 command.c:544 command_print(): debug_level: 3
Debug: 14 1 configuration.c:42 add_script_search_dir(): adding share/openocd/scripts
Debug: 15 1 options.c:96 add_default_dirs(): bindir=/builds/idf/openocd-esp32/openocd-esp32/bin
Debug: 16 1 options.c:97 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/openocd-esp32/share/openocd
[... 373 lines cut ...]
Error: 412 9810 esp32.c:2001 esp32_stub_load(): no working area available, can't alloc space for stub code!
Error: 413 9810 esp32.c:2182 esp32_algo_run(): Failed to load stub (-308)!
Error: 414 9810 esp32.c:2414 esp32_run_do(): Algorithm run failed (-308)!
Error: 415 9810 esp32.c:293 esp32_get_mappings(): Failed to run flasher stub (-308)!
Warn : 416 9810 esp32.c:646 esp32_probe(): Failed to get flash mappings (-308)!
Debug: 417 9810 esp32.c:181 esp32_init_flasher_image(): base=00000000 set=0
Debug: 418 9810 esp32.c:2398 esp32_run_do(): Set arg[0] = 4
Debug: 419 9810 esp32.c:1994 esp32_stub_load(): stub: base 0x0, start 0x40090814, 2 sections
Debug: 420 9810 esp32.c:1998 esp32_stub_load(): addr 0, sz 5891, flags 1
Debug: 421 9810 target.c:1862 alloc_working_area_try_do(): allocated new working area of 5892 bytes at address 0x40090000
Debug: 422 9810 esp32.c:621 xtensa_read_memory(): esp32: xtensa_read_memory: reading 5892 bytes from addr 40090000
Error: 423 9856 esp108_common.c:184 esp108_do_checkdsr(): cpu0: xtensa_read_memory (line 654): DSR (8180CC13) indicates DIR instruction generated an exception!
Warn : 424 9858 esp32.c:659 xtensa_read_memory(): cpu0: Failed reading 5892 bytes at address 0x40090000
Error: 425 9858 esp32.c:2001 esp32_stub_load(): no working area available, can't alloc space for stub code!
Error: 426 9858 esp32.c:2182 esp32_algo_run(): Failed to load stub (-308)!
Error: 427 9858 esp32.c:2414 esp32_run_do(): Algorithm run failed (-308)!
Error: 428 9858 esp32.c:257 esp32_get_size(): Failed to run flasher stub (-308)!
Error: 429 9858 esp32.c:684 esp32_probe(): Failed to probe flash, size 0 KB
Error: 430 9858 core.c:215 get_flash_bank_by_num(): auto_probe failed
Error: 431 9858 gdb_server.c:982 gdb_new_connection(): Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: 432 9858 server.c:100 add_connection(): attempted 'gdb' connection rejected
Info : 433 9861 esp32.c:1651 xtensa_poll(): esp32: Debug controller 1 was reset (pwrstat=0x4F, after clear 0x4F).
Debug: 434 9970 esp32.c:1644 xtensa_poll(): xtensa_poll: core_poweron_mask=3
Debug: 435 9975 esp32.c:1684 xtensa_poll(): xtensa_poll: Core 1 came online, setting up DCR
Debug: 436 9976 esp32.c:886 xtensa_smpbreak_set_core(): xtensa_smpbreak_set_core[esp32] core 1, set smpbreak=230001, state=2`

gerekon, 2019-11-01:
@adexc Could you check it with the latest OpenOCD release? Does this issue still exist?

Onuonga, 2020-08-31:
Any chance someone was able to look into this ? :-)

gerekon, 2020-08-31:
@Onuonga Do you have the same problem with the latest OpenOCD?

gerekon, 2020-08-31:
Since the topic starter did not respond I am going to close this issue unless someone else has the same problem with the [latest OpenOCD release](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20200709).

closed 2021-01-15 by gerekon