espressif/openocd-esp32 #74: The problem with the installation openocd-esp32

DimRusanov, 2019-03-25:
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/flash/nor/xmc4xxx.lo -MD -MP -MF src/flash/nor/.deps/xmc4xxx.Tpo -c src/flash/nor/xmc4xxx.c -o src/flash/nor/xmc4xxx.o
src/flash/nor/xmc4xxx.c: In function ‘xmc4xxx_get_info_command’:
src/flash/nor/xmc4xxx.c:939:51: error: ‘__builtin___snprintf_chk’ output may be truncated before the last format character [-Werror=format-truncation=]
     snprintf(otp_str, sizeof(otp_str) - 1, "- %d\n", i);
                                                   ^
In file included from /usr/include/stdio.h:862:0,
                 from ./src/helper/system.h:25,
                 from ./config.h:338,
                 from src/flash/nor/xmc4xxx.c:19:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 5 and 14 bytes into a destination of size 13
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:3082: recipe for target 'src/flash/nor/xmc4xxx.lo' failed
make[2]: *** [src/flash/nor/xmc4xxx.lo] Error 1
make[2]: выход из каталога «/home/rus/esp/openocd-esp32»
Makefile:4000: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: выход из каталога «/home/rus/esp/openocd-esp32»
Makefile:1722: recipe for target 'all' failed
make: *** [all] Error 2

cd ~/esp/openocd-esp32
./bootstrap
./configure
make  <------What am I doing wrong?
Ubuntu14

igrr, 2019-03-29:
Could you please post the version of GCC you are using to compile? You can get it using `gcc -v` command.

DimRusanov, 2019-03-29:
rus@rus-note:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.3.0-27ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.3.0 (Ubuntu 7.3.0-27ubuntu1~18.04)

XaviDCR92, 2019-03-30:
It looks like a couple of buffers need to be enlarged. I provided a hotfix on the following post: https://esp32.com/viewtopic.php?f=12&t=9901

bas-zach-freeman, 2019-04-17:
I ran into the same issue on Ubuntu 18.04.  I can confirm that the patches mentioned by @XaviDCR92 solved the build issue for me. Thanks.

DimRusanov, 2019-04-19:
Thank you, tried to collect, downloaded the binary version - everything is fine there

-- Dmitry.

gerekon, 2019-11-01:
Fixed in da2fa0f067d8ca65ba896b94bca1a8234640b2b6

closed 2019-11-01 by gerekon