espressif/openocd-esp32 #75: can't install opened-esp32 on MacBook（macos mojave 10.14.4）

cstyle0x007, 2019-04-18:
help！

cd ~/esp

git clone https://github.com/espressif/openocd-esp32

cd openocd-esp32
./bootstrap
./configure
make

build erro。

ransport_tcp.lo libjaylink_la-util.lo libjaylink_la-version.lo libjaylink_la-discovery_usb.lo libjaylink_la-transport_usb.lo  -L/usr/local/Cellar/libusb/1.0.22/lib -lusb-1.0 
Makefile:5465: warning: overriding commands for target `check-recursive'
Makefile:4273: warning: ignoring old commands for target `check-recursive'
restore=: && backupdir=".am$$" && \
	am__cwd=`pwd` && CDPATH="${ZSH_VERSION+.}:" && cd . && \
	rm -rf $backupdir && mkdir $backupdir && \
	if (makeinfo --version) >/dev/null 2>&1; then \
	  for f in doc/openocd.info doc/openocd.info-[0-9] doc/openocd.info-[0-9][0-9] doc/openocd.i[0-9] doc/openocd.i[0-9][0-9]; do \
	    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
	  done; \
	else :; fi && \
	cd "$am__cwd"; \
	if makeinfo   -I doc -I ./doc \
	 -o doc/openocd.info ./doc/openocd.texi; \
	then \
	  rc=0; \
	  CDPATH="${ZSH_VERSION+.}:" && cd .; \
	else \
	  rc=$?; \
	  CDPATH="${ZSH_VERSION+.}:" && cd . && \
	  $restore $backupdir/* `echo "./doc/openocd.info" | sed 's|[^/]*$||'`; \
	fi; \
	rm -rf $backupdir; exit $rc
./doc/openocd.texi:3467: @anchor expected braces.
./doc/openocd.texi:3467: “ {reset_config}
@deffn {Command} reset_c...”扩展起来太长；未扩展.
./doc/openocd.texi:3305: 下一个 reference to nonexistent node `TAP Declaration' (perhaps incorrect sectioning?).
./doc/openocd.texi:3370: 交叉引用 reference to nonexistent node `resetcommand' (perhaps incorrect sectioning?).
./doc/openocd.texi:3285: 交叉引用 reference to nonexistent node `faqrtck' (perhaps incorrect sectioning?).
./doc/openocd.texi:3248: 交叉引用 reference to nonexistent node `targetevents' (perhaps incorrect sectioning?).
./doc/openocd.texi:2282: 交叉引用 reference to nonexistent node `targetcurstate' (perhaps incorrect sectioning?).
./doc/openocd.texi:2189: 交叉引用 reference to nonexistent node `targetevents' (perhaps incorrect sectioning?).
./doc/openocd.texi:2188: 交叉引用 reference to nonexistent node `targetconfiguration' (perhaps incorrect sectioning?).
./doc/openocd.texi:2156: 交叉引用 reference to nonexistent node `gdbportoverride' (perhaps incorrect sectioning?).
./doc/openocd.texi:1951: 交叉引用 reference to nonexistent node `armhardwaretracing' (perhaps incorrect sectioning?).
./doc/openocd.texi:1799: 交叉引用 reference to nonexistent node `usingopenocdsmpwithgdb' (perhaps incorrect sectioning?).
./doc/openocd.texi:1754: 交叉引用 reference to nonexistent node `CPU Configuration' (perhaps incorrect sectioning?).
./doc/openocd.texi:1709: 交叉引用 reference to nonexistent node `TAP Declaration' (perhaps incorrect sectioning?).
./doc/openocd.texi:1390: 交叉引用 reference to nonexistent node `nandconfiguration' (perhaps incorrect sectioning?).
./doc/openocd.texi:1389: 交叉引用 reference to nonexistent node `norconfiguration' (perhaps incorrect sectioning?).
./doc/openocd.texi:1225: 交叉引用 reference to nonexistent node `softwaredebugmessagesandtracing' (perhaps incorrect sectioning?).
./doc/openocd.texi:1058: 交叉引用 reference to nonexistent node `softwaredebugmessagesandtracing' (perhaps incorrect sectioning?).
./doc/openocd.texi:1006: 交叉引用 reference to nonexistent node `autoprobing' (perhaps incorrect sectioning?).
./doc/openocd.texi:888: 交叉引用 reference to nonexistent node `GDB and OpenOCD' (perhaps incorrect sectioning?).
./doc/openocd.texi:795: 交叉引用 reference to nonexistent node `debuglevel' (perhaps incorrect sectioning?).
./doc/openocd.texi:673: 交叉引用 reference to nonexistent node `Tcl Crash Course' (perhaps incorrect sectioning?).
./doc/openocd.texi:94: 菜单 reference to nonexistent node `Command and Driver Index' (perhaps incorrect sectioning?).
./doc/openocd.texi:93: 菜单 reference to nonexistent node `OpenOCD Concept Index' (perhaps incorrect sectioning?).
./doc/openocd.texi:87: 菜单 reference to nonexistent node `License' (perhaps incorrect sectioning?).
./doc/openocd.texi:86: 菜单 reference to nonexistent node `Tcl Crash Course' (perhaps incorrect sectioning?).
./doc/openocd.texi:85: 菜单 reference to nonexistent node `FAQ' (perhaps incorrect sectioning?).
./doc/openocd.texi:84: 菜单 reference to nonexistent node `Tcl Scripting API' (perhaps incorrect sectioning?).
./doc/openocd.texi:83: 菜单 reference to nonexistent node `GDB and OpenOCD' (perhaps incorrect sectioning?).
./doc/openocd.texi:82: 菜单 reference to nonexistent node `TFTP' (perhaps incorrect sectioning?).
./doc/openocd.texi:81: 菜单 reference to nonexistent node `Utility Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:80: 菜单 reference to nonexistent node `Boundary Scan Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:79: 菜单 reference to nonexistent node `JTAG Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:78: 菜单 reference to nonexistent node `Architecture and Core Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:77: 菜单 reference to nonexistent node `General Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:76: 菜单 reference to nonexistent node `PLD/FPGA Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:75: 菜单 reference to nonexistent node `Flash Programming' (perhaps incorrect sectioning?).
./doc/openocd.texi:74: 菜单 reference to nonexistent node `Flash Commands' (perhaps incorrect sectioning?).
./doc/openocd.texi:73: 菜单 reference to nonexistent node `CPU Configuration' (perhaps incorrect sectioning?).
./doc/openocd.texi:72: 菜单 reference to nonexistent node `TAP Declaration' (perhaps incorrect sectioning?).
makeinfo：由于错误删除宏输出文件“doc/openocd.info”；用 --force 保留宏输出。
make[2]: *** [doc/openocd.info] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

gerekon, 2019-05-21:
Hi @cstyle0x007
Could you try `./configure --disable-doxygen-html --disable-doxygen-pdf `?

NavidMitchell, 2019-06-22:
I am having the same error. I tired ./configure --disable-doxygen-html --disable-doxygen-pdf and I still got the same error when running make.

closed 2019-09-27 by igrr