espressif/openocd-esp32 #76: flash_via_openocd

JayNaire, 2019-05-15:
I am using:
- Linux Mint 18
- **eclipse 2019-03**
- openocd-esp32 Open On-Chip Debugger  v0.10.0-esp32-201903 2019-03-13-09:52)
- GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
- ESP_IDF v4.0-dev-512-g6fd535c 12Apr2019
- **Olimex-Arm-Usb-Ocd**
- **Doit esp32 devkit v1 (ESP32-WROOM-32, no on-board ftdi - hence the Olimex )**

I use "make flash" to program the ESP32 via the normal serial usb.
I debug with the Olimex dongle.
It works very well in Eclipse. 

I want to get rid of the serial link completely and just flash and debug via openocd/Olimex, so not using esptool. A manual, test flashing via openocd worked fine but I'd like to integrate with the esp_idf build system.

Before I start re-inventing the wheel:
**Is there something quick and easy like an "*esptool_for_openocd*" or a make target like "*flash_via_openocd*" already in existence?**

Thanks for any pointers!

gerekon, 2019-05-16:
Hi @JayNaire 

> Is there something quick and easy like an "esptool_for_openocd" or a make target like "flash_via_openocd" already in existence?

AFAIK, no.
Also it seems that this issue and respective changes are mostly related to IDF build system, not to openocd.
For now you can use the Makefile rule like the following:

```
OPENOCD_DISTRO = /d/projects/esp/openocd-esp32-win32-0.10.0-esp32-20190313/openocd-esp32

flash_via_openocd:
	$(OPENOCD_DISTRO)/bin/openocd -s $(OPENOCD_DISTRO)/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "init; program_esp32 build/blink.bin 0x10000 verify exit"
```

JayNaire, 2019-05-23:
Hi @gerekon 

Thank you! That works fine and is exactly what I needed.
Much appreciated.

closed 2019-05-23 by JayNaire