espressif/openocd-esp32 #79: GDB skips over code

dschnabel, 2019-06-26:
I am fairly new using gdb on an ESP32 and I am not sure if what I'm seeing is expected. I got debugging working fine using a JTAG and OpenOCD but when I step through the code, some lines are getting skipped. First I thought the problem is that my ELF file is not debug optimized but looking at the compiler output I'm seeing this which tells me that it is debug optimized (-ggdb -Og):
```
GNU C++11 8.2.0 -mlongcalls -ggdb -Og -std=gnu++11 -std=gnu++11 -fno-rtti -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -fexceptions
```

My simple program has these lines:
```
	int a = 1;
	a = 3;
	printf("%d\n", a);
	a = 4;
	printf("%d\n", a);
```
I expect to be able to step through each line, but GDB only stops at the two `printf()` commands. It does not stop at `a = 3;` or `a = 4;`.

Is this a known limitation when debugging the ESP32 or should it be possible to step through each line of code?

Thank you!

igrr, 2019-06-26:
Can you please run `disassemble /m your_function_name` and post the annotated assembly here?

dschnabel, 2019-06-27:
@igrr thank you.

Here is the output:
```
(gdb) disassemble /m app_main
Dump of assembler code for function app_main:
4	{
   0x400d29e4 <+0>:	entry	a1, 32

5		int a = 1;

6		a = 3;

7		printf("%d\n", a);
   0x400d29e7 <+3>:	l32r	a2, 0x400d0348 <_stext+816>
   0x400d29ea <+6>:	movi	a11, 3
   0x400d29ed <+9>:	or	a10, a2, a2
   0x400d29f0 <+12>:	call8	0x400d3290 <printf>

8		a = 4;

9		printf("%d\n", a);
   0x400d29f3 <+15>:	movi.n	a11, 4
   0x400d29f5 <+17>:	mov.n	a10, a2
   0x400d29f7 <+19>:	call8	0x400d3290 <printf>
   0x400d29fa <+22>:	retw.n

End of assembler dump.
```

igrr, 2019-06-27:
This doesn't look unique to the Xtensa toolchain. Here is the same code built with ARM GCC 8.2: https://godbolt.org/z/gFesF-. As you can see, lines with assignments don't have any matching assembly locations, just like the output of `gdb /disassemble` above.

You can, however, get the behavior you need by passing `-O0` flag to the compiler, instead of `-Og` (https://godbolt.org/z/Lqv9Si). In IDF you can add this flag:
- via EXTRA_CFLAGS at project level or CFLAGS at component level for Make build system
- via add_compile_options for CMake build system

dschnabel, 2019-06-28:
Switching from `-Og` to `-O0` did the trick! I can step through all lines now. Really appreciate your help, @igrr 

I assumed `-Og` is equal to `-O0` but it turns out it's not. Also I didn't know about https://godbolt.org/. That's a great tool.

Thanks again. Closing this issue now.

closed 2019-06-28 by dschnabel