espressif/openocd-esp32 #81: esp32_fetch_all_regs error: Segger JLink + ESP32-WROOM

Majoras-Kid, 2019-07-16:
When rewriting the bootloader on a ESP32-WROOM using the Segger JLink an error occurs.

System: Linux Debian x64
Setup: 

1. ESP connected to JLink, power over on-board USB on the esp32
2. Follow steps on [official website](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/jtag-debugging/index.html) to setup OpenOCD (used the current release binary)
3. Write bootloader binary to device using: 
    ```sudo bin/openocd -s share/openocd/scripts -f interface/jlink.cfg -f board/esp-wroom-32.cfg -c "program_esp32 bootloader.bin 0x1000 verify exit"```
    
    3.1 Bootloader binary from esp32 binary build folder```/build/bootloader``` into ~/esp/openocd-esp32
    3.2 Interface setup to be configured for Segger JLink (jlink.cfg)

Error Message:
```
sudo bin/openocd -s share/openocd/scripts -f interface/jlink.cfg -f board/esp-wroom-32.cfg -c "program_esp32 bootloader.bin 0x1000 verify exit"
Open On-Chip Debugger  v0.10.0-esp32-20190708 (2019-07-08-11:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
Info : Configured 2 cores
esp32 interrupt mask on
Info : J-Link V10 compiled Sep  4 2018 11:24:21
Info : Hardware version: 10.10
Info : VTarget = 3.340 V
Info : Reduced speed from 20000 kHz to 15000 kHz (maximum).
Info : Reduced speed from 20000 kHz to 15000 kHz (maximum).
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: cpu0: esp32_fetch_all_regs (line 273): DSR (00000088) indicates DIR instruction generated an overrun!
Error: cpu1: esp32_fetch_all_regs (line 273): DSR (00000088) indicates DIR instruction generated an overrun!
Info : Target halted. PRO_CPU: PC=0x4015B58E (active)    APP_CPU: PC=0x00000000 
Error: cpu0: xtensa_write_memory (line 807): DSR (8000CC13) indicates DIR instruction generated an exception!
Warn : esp32: Failed writing 4 bytes at address 0x3ff5f064, data - a1, 3a, d8, 50, 00, 00, 00, 00
Error: Failed to write ESP32_TIMG0WDT_PROTECT (-4)!
Info : Detected debug stubs @ 3ffb3a3c on core0 of target 'esp32'
Error: cpu0: xtensa_read_memory (line 700): DSR (00000088) indicates DIR instruction generated an overrun!
Warn : cpu0: Failed reading 76 bytes at address 0x50000000
Error: esp32_soc_reset 528 err=-4
in procedure 'program_esp32'

** Unable to reset target **
shutdown command invoked

Info : Restore debug stubs @ 3ffb3a3c on core0 of target 'esp32'
Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()
```

When starting the debugger and trying to halt the ESP32 a similar message can be found (not sure if its relevant/related)
```
telnet localhost 4444Trying ::1...
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> halt
cpu0: esp32_fetch_all_regs (line 273): DSR (00000088) indicates DIR instruction generated an overrun!
cpu1: esp32_fetch_all_regs (line 273): DSR (00000088) indicates DIR instruction generated an overrun!
Target halted. PRO_CPU: PC=0x4015B58E (active)    APP_CPU: PC=0x00000000 
cpu0: xtensa_write_memory (line 807): DSR (8000CC13) indicates DIR instruction generated an exception!
esp32: Failed writing 4 bytes at address 0x3ff5f064, data - a1, 3a, d8, 50, 00, 00, 00, 00
Failed to write ESP32_TIMG0WDT_PROTECT (-4)!
Detected debug stubs @ 3ffb3a3c on core0 of target 'esp32'
esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
xtensa_poll: Target offline
xtensa_poll: Target failure
Polling target esp32 failed, trying to reexamine
xtensa_poll: Target failure
Polling target esp32 failed, trying to reexamine
xtensa_poll: Target failure
Polling target esp32 failed, trying to reexamine
xtensa_poll: Target failure
Polling target esp32 failed, trying to reexamine
esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
> 
```

Does anybody have an idea how to fix this issue? Thanks for answers!

alexdecei, 2019-08-20:
Hi, I have exactly the same error. Any solution ?
I'm actually trying to flash a new firmware via JTAG with a J-Link .

thomasesr, 2019-08-22:
Where can I find documentation about using Generic Segger Jlink interface to upload firmware image o ESP32? I am having the same warning messages:
`Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of irom does not support free_driver_priv()
Warn : Flash driver of drom does not support free_driver_priv()`

alexdecei, 2019-08-23:
The Jlink interface is in openocd-esp32\share\openocd\scripts\interface\jlink.cfg
But I don't know how to configure it or if it need to be configured.

gerekon, 2019-11-01:
> Where can I find documentation about using Generic Segger Jlink interface to upload firmware image o ESP32? I am having the same warning messages:
> `Warn : Flash driver of esp32.flash does not [...]

This is OK.

gerekon, 2019-11-01:
@Majoras-Kid For Jlink you need to lower adapter clock to something like 1000-3000 kHz
```
Info : Reduced speed from 20000 kHz to 15000 kHz (maximum).
Info : Reduced speed from 20000 kHz to 15000 kHz (maximum).
Info : clock speed 20000 kHz
```
In the latest release you can do this by inserting  `adapter_khz 1000` into `interface/jlink.cfg` or by specifying it on the command line like `openocd -c 'adapter_khz 10000' ...`

closed 2019-11-01 by gerekon