espressif/openocd-esp32 #82: Cannot use debugger when 2 Esp-Wrover-Kit attached to PC

brent-afw, 2019-08-23:
Hi,

I recently purchased 2 ESP-Wrover-Kit to facilitate debugging 2 sessions simultaneously using VisualGdb on Windows 10.
The OpenOCD debugger works fine when 1 board is attached at a time. 
However,unfortunately this fails to work when 2 boards are simultaneously attached to the PC. 

I contacted the Sysprogs who in turn suggested I contact Esprissif.

The reply from Sysprogs:- 

"Unfortunately the ESP32-WROVER board does not properly report a unique serial number ("libusb_get_string_descriptor_ascii() failed with LIBUSB_ERROR_INVALID_PARAM" message) and hence OpenOCD cannot select a specific instance of the board by its serial number. Please consider reporting this problem to Espressif and if they provide a fix to OpenOCD that will allow selecting a specific instance of WROVER based on its serial number."

BR,
Brent

igrr, 2019-10-18:
Hi Brent,
It seems the ESP-WROVER-KIT boards don't have any serial number programmed into the EEPROM of FT2232H. This isn't something we can fix in OpenOCD. However you can program the serial number into the board using [FTDI FT_PROG utility](https://www.ftdichip.com/Support/Utilities.htm#FT_Prog) (on Windows) or [ftdi_eeprom utility](https://manpages.debian.org/testing/ftdi-eeprom/ftdi_eeprom.1.en.html) from [libftdi](https://www.intra2net.com/en/developer/libftdi/) (on Linux and macOS).

The config file for ftdi_eeprom would like like this:
```
product="ESP-WROVER-KIT"
vendor_id=0x0403
product_id=0x6010
manufacturer="Espressif Systems"
serial="123456789"  # set to different values for different boards
use_serial=true
```

You can invoke ftdi_eeprom like this:
```
ftdi_eeprom --device i:0x0403:0x6010 --flash-eeprom config
```

brent-afw, 2019-10-19:
Hi Ivan,

Your solution worked.

Thanks
Brent

closed 2019-10-19 by brent-afw