espressif/openocd-esp32 #86: Failed make: cast between incompatible function types

selaliadobor-wt, 2019-09-19:
> **src/target/xtensa_algorithm.c:547:22:**
> **warning: cast between incompatible function types**  from ‘**int (*)(struct target *, struct xtensa_algo_run_data *, uint32_t)**’ to ‘**int (*)(struct target *, struct xtensa_algo_run_data *, void *)**’ [-Wcast-function-type]
>                  **_run->algo_func = (xtensa_algo_func_t)xtensa_run_algorithm_onboard;_**

Currently building with `-Wall` and `-Werror` removed as a work around

fromeijn, 2019-10-21:
Same issue here

gerekon, 2019-11-01:
@selaliadobor-wt  @fromeijn Could you post GCC version number you use?

redmercury, 2019-11-28:
I also get this error.  Compiling on Debian 10 (Buster).
gcc --version = gcc (Debian 8.3.0-6) 8.3.0

WernerMauss, 2019-12-21:
Same issue here. Compiling on Raspbian 10 (Buster). 
gcc --version = gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

WernerMauss, 2019-12-21:
> Currently building with `-Wall` and `-Werror` removed as a work around

How did you remove them?

gerekon, 2019-12-27:
Ok, reproduced. Will be fixed soon. Thanks, guys!

gerekon, 2020-02-19:
Closed with 4ce5744e

closed 2020-02-19 by gerekon