espressif/openocd-esp32 #88: Build erro in mflash.c

petrkr, 2019-12-17:
Can not build on Arch linux

I tested also older tags (back to March 2019)

```
In function ‘mg_gen_ataid’,
    inlined from ‘mg_storage_config’ at src/flash/mflash.c:1174:2:
src/flash/mflash.c:1162:2: error: ‘memset’ offset [509, 512] from the object at ‘buff’ is out of the bounds of referenced subobject ‘reserved7’ with type ‘mg_io_uint8[186]’ {aka ‘unsigned char[186]’} at offset 322 [-Werror=array-bounds]
 1162 |  memset(pSegIdDrvInfo->reserved7, 0x00, 190);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3355: src/flash/mflash.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
```

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
gcc version 9.2.0 (GCC) 
```

petrkr, 2019-12-17:
Found workaround how to build it:

Use configure with `--disable-werror` parameter
`./configure --disable-werror`

But it does not mean it can be closed :)

gerekon, 2020-01-18:
What GCC version do you use?

erhankur, 2023-01-17:
@petrkr we are in sync with the mainline OpenOCD repo. If you still have an issue please create a ticket in https://sourceforge.net/p/openocd/tickets/

closed 2023-01-17 by erhankur