espressif/openocd-esp32 #89: Unable to debug ESP32-WROVER unless "Flash SPI Speed" set to 20MHz

oliverjc, 2019-12-22:
**JTAG Interface:** ESP-PROG
**MCU:** LyraT 4.3

After much trial and error, I managed to successfully get the ESP-PROG interfacing with my Lyra-T development board.  I kept on getting `Receiving JTAG scan chain interrogation failed: all ones` (or zeros) error similar to #35. I tried everything from different cables, connectors, microcontrollers, but that fix, in the end, was to set the SPI Flash Speed to 20MHz. 

This wasn't anywhere in the documentation so there's a chance it could be a bug, if not, it's definitely worth adding to the [Tips & Quirks Documentation](https://docs.espressif.com/projects/esp-idf/en/v3.2.3/api-guides/jtag-debugging/tips-and-quirks.html).
I'd be happy to put in a pull request for the Docs if it seems like a reasonable thing to do.

gerekon, 2020-01-19:
Hmm..., have had such problem. Have you tried with lower JTAG freq, w/o lowering SPI flash speed?

erhankur, 2023-01-17:
Closed due to inactivity.

closed 2023-01-17 by erhankur