espressif/openocd-esp32 #90: esp32: Debug controller was NOT reset

techtoys, 2020-01-01:
JTAG adapter type: FT2232H wire up with ESP-WROOM-32D
Release of ESP-IDF: Ver 4.1
Details of OS debugging: Windows 7 Pro 64-bit on a PC
Pinout: 
GPIO15=TDO(ADBUS2)
GPIO12=TDI(ADBUS1)
GPIO13=TCK(ADBUS0)
GPIO14=TMS(ADBUS3)
EN=ADBUS5

Issue: After `JTAG tap: esp32.cpu0 tap/device found and esp32.cpu1 tap/device found` , the expected message `esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F)` is **not found**. Instead, it goes on with `JTAG tap: auto0.tap tap/device found: 0x120034e5...` until the end an error message as :
`Error: IR capture error at bit 105, saw 0x4210842108....`

Attached please find the openocd_log.txt.
Unfortunately, when I tried to generate the gdb_log.txt file with `xtensa-esp32-elf-gdb -ex "set remotelogfile gdb_log.txt" gdbinit build/blink.elf"` I have got this error from command prompt:

C:\esp-idf-4.1\eamples\get-started\blink/gdbinit": not in execuatable format: File format not recognized

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/4013810/openocd_log.txt)

igrr, 2020-01-01:
@techtoys please try reducing the adaptor clock frequency.
GDB log is not required, as the issue is with communication between OpenOCD and the target. But a log from OpenOCD with `-d 3` command line flag would be useful.

techtoys, 2020-01-01:
@igrr I tried a slow adapter clock frequency 2Mhz, 1MHz, and 100kHz with the same result. 
Log with `-d 3` command line flag is attached. 
I am starting to think if the issue is a hardware problem. Please notice that it is not the official ESP-WROVER-KIT. I have built a board with FT2232H + ESP-WROOM-32D + graphical controller on a single board. No AT93C46 has been soldered as I think FT2232 is all software configurable. Please advise if my thinking is correct.

The screenshots from command line also attached.
[image]
[image]

[openocd.log](https://github.com/espressif/openocd-esp32/files/4013932/openocd.log)

techtoys, 2020-01-01:
An observation on `esp32_devkitj_v1`. 
```
## TCK, TDI, TDO, TMS: ADBUS0-3
## LEDs: ACBUS4-7
ftdi_layout_init 0x0008 0xf00b
```
Question:
Since TDI is an input at ADBUS1 with TMS(ADBUS3):TDO(ADBUS2):TDI(ADBUS1):TCK(ADBUS0), should it be `ftdi_layout_init 0x0008 0xf00d` instead?

techtoys, 2020-01-03:
Do I need to enable JTAG debugging in ESP-WROOM-32 to start communicate openOCD with ESP?
I have there is an option in Menuconfig > Security features > Potentially insecure options > Allow JTAG Debugging. Not sure what it is.

igrr, 2020-01-03:
No special configuration is required on the ESP32 side. Just the application should not use GPIOs 12-15 for other purposes (such as SPI, SD, etc).

Enabling secure boot will also permanently disable JTAG debugging by burning an eFuse, but this can be overridden using the option you found under "Potentially insecure options".

techtoys, 2020-01-04:
@igrr Do you have a logic analyzer to check what happens on the JTAG interface, and decode the transfers?

Yes, I have a logic analyzer. Before this step I have tried several attempts:
1) Uninstall IDF 4.1 and reinstall IDF 3.2.3 and start everything fresh.
2) Make use of the blinky example in `\examples\get-started\blink`, I rebuild the whole project with the menuconfig option [Enable flash encryption on boot]>[Allow potentially insecure options]>[Potentially insecure options]--->[Allow JTAG Debugging].
3) <Exit> and rebuild the whole project with `idf.py all`. Project build is successful without error.
4) Finally, I tried to flash the WROOM-32 with `idf.py -p COM84 flash`  I have got an error as `Detected overlap at address: 0x8000 for file: partition_table/partition-table.bin`.

Normal build **without** [Allow JTAG Debugging] turning on from "Potentially insecure options" was OK, and flash also OK. 

Why is it so?

igrr, 2020-01-04:
The bootloader size was probably increased when you have enabled the "Enable flash encryption on boot" option. Please make sure you read and understand the secure boot documentation before enabling this option, as incorrect usage may permanently brick the chip. If you are debugging JTAG functionality, it is very likely that you don't need to touch secure boot related options.
From the log you have provided, it can be seen that the ESP32's JTAG TAPs are found correctly, but then OpenOCD doesn't detect the end of the JTAG chain. I think this may be due to incorrect physical connections between the debug adapter and the ESP32. Do you happen to have a known good adapter (Wrover-kit, or JLINK) to verify the hardware connections?

techtoys, 2020-01-05:
@igrr Do you happen to have a known good adapter (Wrover-kit, or JLINK) to verify the hardware connections

I happen to have a TI's CC3200 launchpad with a FT2232D onboard. So I changed the ftdi_vid_pid to 0x0451 0xc32a to fit the -d version instead of -h version of FT2232. 
One more modification required: ftdi_layout_init 0x0008 0x000b ---> 0x0008 0x004b because there is a gate to control JTAG on/off from ADBUS6 pin. 

Wiring the boards from JTAG pins (only TMS, TDO, TCK, & TDI) now I can see esp32 is connected and reset successfully. Still have to try program stepping but it is a good sight with FT2232D.

Therefore, the issue was due to some mistake with the FT2232HL circuit. I will double-check. 

Thanks a lot for your help.
[image]

closed 2020-01-05 by techtoys