espressif/openocd-esp32 #91: esp32 open-ocd does not connect to esp32-wroom-kit

rebeliousconformist, 2020-01-08:
directory is 
```
espressif/openocd-esp32/tcl
```

openocd -f board/esp32-wrover-kit-1.8v.cfg gives the following output

```
Open On-Chip Debugger  v0.10.0-esp32-20191114-1083-g97ba3a6b (2020-01-08-15:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'

Warn : Flash driver of esp32.flash does not support free_driver_priv()
Warn : Flash driver of esp32.irom does not support free_driver_priv()
Warn : Flash driver of esp32.drom does not support free_driver_priv()
````

lsusb gives
```
<pre>Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:5409 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
</pre>
```

gerekon, 2020-01-18:
Could you try `openocd -f board/esp32-wrover-kit-3.3v.cfg`?

rebeliousconformist, 2020-01-20:
Hi @gerekon yes i tried with it. 
So i also tried hooking up  the esp32-wroom-32module (custom HW board design) to a jtag adapter.
The WROOM module DOES NOT WORK with a WROOM script. The Custom WROOM module works with a the above mentioned WROVER script . 
i can debug and i am satisfied with the performance. But there seems to be a discrepancy in the openocd script.
Regards.

gerekon, 2020-01-20:
@rebeliousconformist 
> The WROOM module DOES NOT WORK with a WROOM script. 

[esp32-wroom-32.cfg](https://github.com/espressif/openocd-esp32/blob/master/tcl/board/esp-wroom-32.cfg#L1) is deprecated and does not include JTAG interface config. To use it you need to run `openocd -s espressif/openocd-esp32/tcl -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wroom-32.cfg`. 

> The Custom WROOM module works with a the above mentioned WROVER script .

AFAIK ESP-WROOM-32 need 3.3V flash voltage so you need to use `board/esp32-wrover-kit-3.3v.cfg`

sdrabb, 2020-06-01:
- Create the following file: `/etc/udev/rules.d/esp32link.rules `
- Containing these lines:
- `ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="664", GROUP="plugdev"`
- Restart your computer

closed 2020-06-17 by rebeliousconformist