espressif/openocd-esp32 #92: Path error with program_esp

techtoys, 2020-01-12:
Environment: MSYS32 
OS: Windows 7 64-bit
Board: ESP-WROOM-32D
OPENOCD: FT2232H

Now I have setup openocd-eps32 with ESP-WROOM-32.cfg and it is working OK. In MSYS2 openocd is listening at port 3333 with this command
`openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg `.

Then I tried to program the board with command below:
`openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp build/blink.bin 0x10000 verify exit"`.
This command works only if I go into the directory of the blink project; i.e. the location at `IDF_PATH/examples/get-started/blink`. Everything works in Command-Line.

The next step to try is the Eclipse IDE with commands at the beginning of “Initialization Commands” by following the ESP-IDF Programming Guide:
`mon program_esp ${workspace_loc:blink/build/blink.bin} 0x10000 verify`.

Error message from MSYS2 openocd shell is :
`** Programming Started **
auto erase enabled
Error: couldn't open C:msys32homeJohnespesp-idfexamplesget-startelinuillink.bin
embedded:startup.tcl:480: Error: ** Programming Failed **`

It seems Eclipse was able to locate the IDF_PATH at `C:/msys32/home/John/esp/esp-idf/examples` but it failed to concatenate with the directory at `blink/build/` to locate the binary file `blink.bin`.

How to solve it?

gerekon, 2020-01-18:
> openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg -c "program_esp build/blink.bin 0x10000 verify exit". This command works only if I go into the directory of the blink project; i.e. the location at IDF_PATH/examples/get-started/blink

Yes, because you specifies relative path to app binary `build/blink.bin`. If you want to use this command from any location, please, specify absolute path to binary.

> The next step to try is the Eclipse IDE with commands at the beginning of “Initialization Commands” by following the ESP-IDF Programming Guide:
mon program_esp ${workspace_loc:blink/build/blink.bin} 0x10000 verify

Probably this happens because you run Eclipse under Windows and `workspace_loc` uses backslashes. Could you try absolute path `C:/msys32/home/John/esp/esp-idf/examples/get-started/blink/build/blink.bin`? Or try to run OpenOCD in Windows console instead of MSYS2. Note that ESP-IDF drops support for MSYS environment.

techtoys, 2020-01-20:
> @gerekon try absolute path
Yes, using the absolute path it is working, even in MSYS environment. 
Thanks a lot.

closed 2020-01-20 by techtoys