espressif/openocd-esp32 #94: py_debug_backend repository doesn't exist

mbilloo, 2020-02-08:
When checking out master and running ./bootstrap, checkout of https://github.com/espressif/py_debug_backend.git/ during submodule checkout fails because the repo doesn't exist.

XaviDCR92, 2020-02-13:
I can confirm this issue as well. I have tried removing the entry from `.gitmodules`:

`
[submodule "testing/esp/py_debug_backend"]
	path = testing/esp/py_debug_backend
	url = ../../espressif/py_debug_backend.git
`
Then I ran `git submodule sync` with no luck. For some reason, `py_debug_backend` is still being referrenced on `.git/index`.

mbilloo, 2020-02-13:
This will require a PR since the reference to the submodule will need to be removed from the index (https://stackoverflow.com/questions/1260748/how-do-i-remove-a-submodule#1260982).

gerekon, 2020-02-18:
Yes. We have not published that repo yet. It is used only for running our tests. We will remove that reference soon.

closed 2020-02-18 by espressif-bot